jungmannlab/picasso #1: Cannot undrift using RCC

shalinshah1993, 2017-07-12:
Hello, I collected 8000 frames at 500 ms and I was trying to drift correct. I get following error:

[image]

I tried to reduce the number of segments from 1000 to 100 but then I get a different type of error: 

[image]

Sorry for uploading images. GitHub's issue tracker didn't allow me to copy text from dialog box.

straussmaximilian, 2017-07-12:
Hi Shalin,
thanks for letting us know.
The first kind of error usually arises when there is too much drift between the slices and the fitting gets ill-conditioned. However, when decreasing the slice side too much there are not enough events for the RCC. How many events do you have on average in each frame? You could try to increase the slices or start with unfrift from picked - you could still try the RCC afterward.

I'll update the error messages for the next release and probably include a more robust fitter.

Best,
Maximilian

shalinshah1993, 2017-07-12:
I tried to increase the number of segments to 4000 and this is what I get.

[image]

In general, I think there are some frames when there is no localization. In general, I recorded at 500 ms exposure and imager strands were of length 8. I see 1 - 3 extremely bright blinks on an average per frame and a few dull spots which I assume are nonspecific binding. 

If I perform undrift from picked, I assumed it had to be on for the entire time to work.  I haven't used any markers for drift correction and I read about quantum dots/ gold nano-particles in your nature protocol paper. Does gold nano-particle work with emission range of ATTO 488 and ATTO 647N?

straussmaximilian, 2017-07-12:
For 1-3 bright spots per frame, most of the algorithms won't work.  I would recommend having at least 50 per frame. The last acquisition I took had around 400. Undrift from picked will also work if spots are not on permanently and you have blinking - given you have enough events per frame. The AuNp should work with that emission range. You could also try a DNA-origami structure with a lot of handles on it.

shalinshah1993, 2017-07-12:
Alright, sounds good. I will re-collect data and probably add nanoparticles or origami with relatively long PAINT handles.

closed 2017-07-13 by straussmaximilian