jungmannlab/picasso #4: RCC Drift correction fails

preynoldssco, 2018-03-26:
[image]

After fitting, I can't apply drift correction in Render module.
Should I be setting up picks or segmentation before running drift correction?

locs file:
https://gla-my.sharepoint.com/:u:/g/personal/paul_reynolds_glasgow_ac_uk/EYVpKeE6fT9MrINQAnNJ2zoBlCqp9aK2tiwejKF03A12DQ?e=Enbg9M

win10, in anaconda environment as in readme.

straussmaximilian, 2018-03-26:
Hi,
I tried to reproduce the error but unfortunately wasn't able to reproduce on Linux and Windows.
[image]

The recommended approach is to use RCC with default settings (1000). If it should fail adjust (usually increase) with respect to number of locs/frame in order to be able to correlate. Then pick markers (e.g. beads / Au-Np or DNA-origami) and use undrift by pick.

Which segmentation parameter did you choose?
From the error message it looks like the progress bar is failing to initialize, but that should not even be the case if the segmentation parameter is wrongly chosen.

preynoldssco, 2018-03-26:
Looks good on your system!

I've tried small and large segmentation parameter values with the same error. (1,500,1000,5000,10000,20000,50000) - so I think it's failing before that.

I'll rebuild the environment again? It's possible I've made a mistake along the way, but I've had this same error on two machines.

straussmaximilian, 2018-03-26:
Hmm. At this point I can only imagine some package incompatibility.
Can you quickly verify your tqdm version? Should be 4.18.8. 
Try conda list:
[image]

You can try to install the tqdm package again:
conda: conda install -c conda-forge tqdm or
pip: pip install tqdm

Let me know whether that helped. Thanks.

preynoldssco, 2018-03-26:
That's the one - updated tqdm and that's fixed it thanks! Drift correction now OK, progress bar and all

For reference, this was my package list after my attempt at setting things up:
[image]

straussmaximilian, 2018-03-26:
Great! Happy that it works now. Will place a little note in the readme.

closed 2018-03-26 by straussmaximilian

preynoldssco, 2018-03-26:
Thanks

Another small edit for the readme is that activating the environment is different on windows/mac

Mac:
source activate picasso

Windows:
activate picasso

straussmaximilian, 2018-03-26:
Will change. Thanks