jungmannlab/picasso #6: Wish-List

straussmaximilian, 2018-03-30:
Current (pot. incomplete) wish-list:

## Nice to have
- 2D/3D Implementation for Phasor
- Importer/Exporter for Thunderstorm
- Implement HDBSCAN for Clustering
- Create a model from existing data for simulation optimization
- Open File resets Window
- Support for non ome-tif
- Remember last working directory for all modules
- Slider for Localize and better arrow functions (e.g. shift + arrow moves 10 frames)

## Room for improvement
- Update from PyQt4 to PyQt5
- Speed up 3D display
- Work on documentation
## Known bugs:

- Freezing MLE Fits
- Fitting Erros in Mac Os X
- RCC can fail

Feel free to comment and add or open a new issue. I'll try to keep this list up to date.

cleterrier, 2018-03-31:
Sorry didn't see this before opening issue on ThunderSTORM export.

A small thing that would be nice is to have the settings of localize/render etc remembered between sessions (even from days to days). At least the camera parameters could be stored in a file because they are not likely to change from one day to another for most people.

Another thing: A way in the GUI to process just a part of a raw sequence (a subset of frames with start/stop boxes) for testing.

cleterrier, 2018-03-31:
Great for the camera, will look into it. For the partial processing, I was thinking simple start/stop frame boxes. It can be useful to process part of the whole sequence, for testing. Another use case is when you want to exclude some frames at the begining (like 100) because initial blinking is too dense, which occurs sometimes in STORM.

joschnitzbauer, 2018-04-02:
@cleterrier this type of filtering is conveniently done with “Picasso Filter”. Open the locs file, select column header “frame”, make a histogram and click-drag the region of the histogram you want to keep.
you still have to Localize in frames you might not be interested in, but that hopefully does not take long.
You can also filter in two dimensions at the same time with a 2d histogram (click on two column headers while pressing ctrl), and make a 2d histogram.
Hope this helps!

cleterrier, 2018-04-02:
Thanks for the information, I will try that. Is there a way to filter precisely at defined values? It's usually difficult to do by clicking on an histogram.

Once one knows precisely what type of filtering is wanted, having an expression box allowing to rapidly filter on multiple conditions, with precise values would be great. There is something like this in ThunderSTORM, you can do all your cleanup in one step with something like "photons>1000 & photons<100000 & length<10" (where "photons" and "length" are column headers). Maybe something like that could be added to the render module, similarly to the expression box in the localize module?

straussmaximilian, 2018-04-02:
I like the idea of having a a precise filter with exact settings. I'll ad this to the wishlist.

cleterrier, 2018-04-02:
Cool, thanks!

cleterrier, 2018-04-02:
I'd like to add an item to the wish-list: correction of drift in Z.

Even with moderate length acquisitions (20 minutes, relatively to PAINT acquisitions that can last hours), I already see some Z drift in the order of 50-100 nm. This is with optical focus stabilization, a Nikon PFS that is performing very well. This is significantly degrading the resulting reconstruction when viewed in XZ sections. I know because N-STORM (Nikon software) does the Z correction so I can check with and without it.

Regarding open-source Python implementations, Hazen Babcock storm analysis suite (from Xiaowei Zhuang's lab) does Z drift correction. Hazen Python's code is here: https://github.com/ZhuangLab/storm-analysis/blob/master/storm_analysis/sa_utilities/xyz_drift_correction.py

I don't think Z drift can be corrected by RCC, it would be too noisy for that (not enough points). So a simple autocorrelation step in Z after the XY correction by RCC may be the optimal solution.