jungmannlab/picasso #7: Size or frame limit for movies?

preynoldssco, 2018-03-31:
My data is in large tif files, e.g. 50,000 frames, 256x256. That's a 6.4Gb file, which only loads frame 1 in localize.

I truncated it to 500 frames and 30,000 frames and these open fine. Do you know of a size limit imposed by the way movies are handled in the code? I can try other movie sizes if that helps.

cleterrier, 2018-03-31:
This is due to the way ImageJ encodes tiff files larger than 4 GB. They are not real tiff files, rather raw files with a small header. Almost no software understands this format (libtiff doesn't). What is possible is to use the ImageJ plugin in this repository (the jar here: https://github.com/jungmannlab/imagej-raw-yaml-export/tree/master/target) to save large tif stacks as .raw + .yaml and process the raw file with Picasso.

cleterrier, 2018-03-31:
Looks like we were writing the same thing simultaneously!

preynoldssco, 2018-03-31:
Thanks both - I'll switch to raw format for now (my head can't handle having movies split into multiple sections)

alexauer, 2018-03-31:
I've updated the ImageJ plugin (commit 12) for the raw yaml exporting. The compiled jar file can now be found in the folder `jar` located in the root directory of the project.

joschnitzbauer, 2018-04-01:
Just chiming in here with a note: Localize is very inefficient with big endian byte order (so bad I should not have even allowed them to be opened). But there is a quick way to save any raw file from ImageJ with little endian byte order. The setting in ImageJ is here:

Check "Save TIFF and Raw in Intel Byte Order" in the Edit>Options>Input/Output dialog box to save using little-endian byte order.

Saving then in ImageJ as raw file is very fast and loading it into Localize as well.
When loading into Localize, a dialog will open where you can specify file information:
- Byte order
- Data type (uint16 in most cases)
- Number of frames 
- FOV Width
- FOV Height

After loading, Localize will produce a .yaml file with all this information, so that you don’t have to type it in the next time.

This is by far the fastest and quickest way of importing and processing movies from other file formats.

Cheers!
/J

cleterrier, 2018-04-01:
just a thought (but really it may be stupid): could it be a strategy to incorporate the BioFormat library for import/export of image file formats? I don't know if they can be used from Python or if it would make the install a lot bigger or more difficult.

preynoldssco, 2018-04-01:
BioFormats would make it more accessible. For example my data is from a Zeiss setup and therefore in .czi to begin with.

However, batch converting to .raw seems like the most efficient way to go. Already impressed with file handling speed in Picasso - well done devs

joschnitzbauer, 2018-04-01:
Agreed, supporting more file formats natively would be fantastic, but it might be a beast to maintain.
Especially when you think of a, in my opinion, key feature of Localize: lazy memory-loading of frames as they are needed. This allows the user to see the movie instantly when loading it. It’s quite custom code to support this for each file format.
If other file formats are implemented without this feature, I’d suggest that Picasso is at least verbose about the fact that the user’s workflow is not optimal.

preynoldssco, 2018-04-01:
Converted my data to .raw using the plugin from Alex, but I see the same behavior in Localize.

File is 6.4Gb and opens only frame 1. No dialogue box etc in Localize.

Converted to .raw again using plugin with "Save TIFF and Raw in Intel Byte Order" checked. Same issue with just one frame loading. No dialogue box, image is now white noise.

joschnitzbauer, 2018-04-01:
@preynoldssco could it be that there is a yaml file leftover from your try with the ImageJ plugin? Try deleting that before loading the “intel byte order”-file.
(If there is a yaml file already, Picasso takes it as a given and does not present you with the dialog)

alexauer, 2018-04-01:
@preynoldssco does the yaml file created with the ImageJ plugin state `Frames: 1`? I've found a annoying circumstance in ImageJ. ImageJ treats stacks differently depending how they are imported. Bioformats imports a stack as time points (frames). The IJ function "make substack" creates a stack of slices (z slices). Depending on the data typ the plugin saved a wrong number of frames usually `1`. I've tested the plugin now with bioformats imported zeiss data .czi and make-substack created .tiff data.
Lazy load for Localize should now also be supported with the raw and yaml files exported from the plugin.

preynoldssco, 2018-04-01:
Thanks both, I'm making this harder than it has to be!

@joschnitzbauer  you were right, there was an existing yaml file getting in the way, that's an easy fix

@alexauer that's a nasty little quirk in ImageJ. I usually open my .czi as a virtual stack and then save in desired format. That gives me the `Frames: 1` problem. Duplicating that stack also gives the 'Frames: 1' problem. If I make a substack, I need to save it as .tiff first, otherwise the system path is null and the imageJ plugin bails.  It's time consuming to save 60k frames as .tiff again after already waiting for the substack.

Here's that error:
[image]

And so back to @joschnitzbauer - I think the raw export plugin is trying to catch the null path of a file that hasn't been saved already, but misses it? [Looking at the code](https://github.com/jungmannlab/imagej-raw-yaml-export/blob/50b6be547d1942a64f6b1e0175c209c0532d921e/src/main/java/org/jungmannlab/imagej/Raw_Yaml_export.java#L77), it should request a path for the raw export - but that isn't happening as above

preynoldssco, 2018-04-01:
Looks like I had an old .jar in my plugin folder. New .jar you uploaded today works fine @joschnitzbauer 
Plays well with batch conversion macro in ImageJ as well

joschnitzbauer, 2018-04-02:
@preynoldssco glad it worked!
Just for the record: @alexauer is the author of the ImageJ plugin who fixed this. :)

closed 2018-10-25 by straussmaximilian