jungmannlab/picasso #8: Export as ThunderSTORM format

cleterrier, 2018-03-31:
It would be great to have an option in the render module to export as ThunderSTORM csv format. I detailed the format of the columns here: http://www.neurocytolab.org/tscolumns/

Most important columns to be able to use ThunderSTORM as a filtering/rendering engine downstream of Picasso are:
x [nm],y [nm], z [nm], intensity [photon], uncertainty_xy [nm], uncertainty_z [nm], detections

I'm not sure if Picasso does calculate z uncertainties, if not that column can be omitted.

Right now there is no text format exported by Picasso that has the xy uncertainty (they are not in the NIS or the FRC exported files), so it's not possible to render the resulting localizations with a Gaussian reconstruction.

Alternatively (or in addition), some documentation about the localization file format used by Picasso (how is the data organized within the hdf5 file) would be useful.

straussmaximilian, 2018-04-03:
Feel free to pull the dev branch: https://github.com/jungmannlab/picasso/tree/dev.
I have implemented an exporter for thudnerSTORM-csv in render as well a converter for csv2hdf in command line. See:
https://github.com/jungmannlab/picasso/wiki/Render#export-as-csv-for-thunderstorm
https://github.com/jungmannlab/picasso/wiki/CMD#csv2hdf

Support for linked files (column detections) still needs to be implemented. z uncertainty is calculated from the lpx and lpy (sqrt of CRLBs).

cleterrier, 2018-04-04:
Just a detail: there is a space added in at the end of the name for the "uncertainty_xy [nm]" column in the picasso exported files. I have scripts that detects columns to manipulate them and they failed, that's how I tracked it down...

straussmaximilian, 2018-04-04:
I removed the space with commit #964.

cleterrier, 2018-04-06:
Works better with my script - however ThunderSTORM still complains about some "null" it does not understand when trying to link localizations. It's not a problem if we can link in PIcasso and export the resulting "length" or "n" columns in Picasso file as "detections" in the csv file - any chance we could get that?

Also, I have this warning when exporting a csv from Render:
C:\Users\NeuroCyto\Anaconda3\envs\picasso\lib\site-packages\numpy\core\records.py:507: FutureWarning: Numpy has detected that you may be viewing or writing to an array returned by selecting multiple fields in a structured array.
This code may break in numpy 1.13 because this will return a view instead of a copy -- see release notes for details.
  return obj.view(dtype=(self.dtype.type, obj.dtype))

straussmaximilian, 2018-04-06:
I implemented the collections import with commit #969. I chose the `len` parameter - that's duration of an event and not the number `n` of linked localizations. I imagined that this probably makes more sense for kinetic analysis. For a gradient that is well chosen `n` should be `len` anyhow. Additionally, for a gap size of 0 `len = n`.

I noticed that a lot of warnings and unnecessary output is displayed when using the terminal, I'll add a revision note to the wish-list. Unfortunately, there are many that cannot be solved immediately (i.e., when using deprecated packages where I would need to check compatibility first.  Potentially it also makes sense to continuously upload the current installer (it's not using the terminal).

cleterrier, 2018-04-07:
Exporter works great. I could add Picasso to my software comparison (see here https://twitter.com/christlet/status/982370186362871808?s=19) where I use each software for 3D calibration and localization, then ThunderSTORM for rendering.

straussmaximilian, 2018-04-07:
Great, that's awesome. I just updated the wiki with the Exporter Specifications and close this now.

cleterrier, 2018-04-09:
One thing I realized this week-end while working on a trasnlator is that ThunderSTORM columns headers actually have "brackets" around the names, i.e. 
x [nm]
is actually
"x [nm]"
in the produced csv file.
Not sure if this is intended (it works without them) but if you want to stick as closely as possible to the TS format could be useful to add them

cleterrier, 2018-04-09:
Also I chased the problem with the merging csv files from Picasso  within ThunderSTORM. If you want to merge (link localizations in subsequent frames) in ThunderSTORM a file from Picasso, you get this error:
`Cannot update then parameters: null`

This is because the csv file exported by Picasso lacks the "bkgstd [photon]" column, i.e. the standard deviation of the background (the value used to calculate the uncertainty in the Mortensen formula). If I add a mock bkgstd column in a Picasso csv file I can merge in thunderSTORM (although it will complain, understandably, that it lacks some data such ads the ADU to photons factor to correctly recalculate the uncertainties).

straussmaximilian, 2018-04-09:
I will change the columns to include the brackets in order to stick as closely to the TS format.

As for the bkgstd [photon] column, I didn't include it as we don't save it in the columns. 
If it's causing problems in downstream processing, I'll potentially start to include it in the *.hdf5 files. Just have to figure out how to make everything backward compatible.

closed 2018-10-25 by straussmaximilian