jungmannlab/picasso #12: toraw fails on standard tif

preynoldssco, 2018-04-01:
I'm assuming it's not designed to work with .tif files that aren't .ome.tif, but I'm hitting an error if I try to open standard .tif stacks in toraw

[In io.py](https://github.com/jungmannlab/picasso/blob/2af14651af774cdf0d03878b3de28cfa9f53afdf/picasso/io.py#L420) the file structure has .ome.tif hard coded in:
` pattern = _re.compile('(.*?)(_(\d*))?.ome.tif')`

Which results (I think?) in this error when trying to set up a batch convert to raw:
[image]

You may not see this as an issue? I should probably be storing data in ome.tif rather than in .czi > .tif or .raw

joschnitzbauer, 2018-04-02:
“To raw” was specifically designed to convert and concatenate tifs from MicroManager.
Because tifs have a size limit of 4GB, MM splits a long movie into multiple files with a certain naming convention. This convention is being caught with that regex expression.
So, “to raw” is not meant to batch-convert many files even though it looks like it (program description should have been more precise here...)
ImageJ and @alexauer’s plugin are probably the best way to go right now for converting non-MM movies.

preynoldssco, 2018-04-02:
OK, thanks  - I'll keep that in mind if we ever build our dedicated STORM rig and run it from MM.....

closed 2018-04-02 by preynoldssco