jungmannlab/picasso #14: Wishlist: Z drift correction

cleterrier, 2018-04-02:
I'd like to add an item to the wish-list: correction of drift in Z.

Even with moderate length acquisitions (20 minutes, relatively to PAINT acquisitions that can last hours), I already see some Z drift in the order of 50-100 nm. This is with optical focus stabilization, a Nikon PFS that is performing very well. This is significantly degrading the resulting reconstruction when viewed in XZ sections. I know because N-STORM (Nikon software) does the Z correction so I can check with and without it.

Regarding open-source Python implementations, Hazen Babcock storm analysis suite (from Xiaowei Zhuang's lab) does Z drift correction. Hazen Python's code is here: https://github.com/ZhuangLab/storm-analysis/blob/master/storm_analysis/sa_utilities/xyz_drift_correction.py

I don't think Z drift can be corrected by RCC, it would be too noisy for that (not enough points). So a simple autocorrelation step in Z after the XY correction by RCC may be the optimal solution.

straussmaximilian, 2018-04-02:
Currently, you can use "Undrift from Picked (3D)" if you have fiducial markers (i.e., AuNP). I'll check out Hazens Code. As for the RCC in z, an implementation should be straightforward (only flip the axis), so I potentially give it a try.

cleterrier, 2018-04-03:
Regarding Z correction, I'm not sure if flipping the axis would work (or maybe I don't understand what you mena by that. If so, sorry). When you do XY drift correction you are projecting the Z information, which is OK as it's not a lot of information compared to the amount in XY, so you keep most of the info for correlation. If you flip one axis and process the image in an XZ plane for example, you will project the Y dimension and loose a lot of information. Correlating reconstructed XZ projections will likely be very noisy.

One possibility is to cross-correlate in 3 dimensions at once, I think this is what Hazen is doing. But then doing it redundantly (RCC) is likely to become computationally challenging. The right option IMO is to first do RCC for XY drift correction, then use the XY drift-corrected coordinates to build sub-images in 3D, and correct the Z drift by cross-correlation of these sub-volumes only in the Z direction.

rafalkowalewski1, 2024-07-05:
In Picasso 0.7.0 AIM (https://www.science.org/doi/10.1126/sciadv.adm7765) will be implemented. This works great for 3D data too.

Best wishes,
Rafal

closed 2024-07-05 by rafalkowalewski1