jungmannlab/picasso #15: Wishlist: Chromatic aberration correction

cleterrier, 2018-04-02:
I'd like to add an item to the wishlist: chromatic aberration correction.

When recording multiple channels with spectrally different fluorophores, correction of the chromatic aberrations is necessary to register the channels at the nanoscale. For example, we usually do 2-color PAINT in red and far-red channels simultaneously (561 and 647 lasers, alternated frames) and warp the channels to register them.

Several software do this, using either polynomial warping (N-STORM does this) or B-Splines (like DoM Utrecht https://github.com/ekatrukha/DoM_Utrecht/wiki/Chromatic-correction). A Matlab implementation exists in SharpVisu (https://github.com/andronovl/SharpViSu). There may be a Python implementation in PyMe but I'm not sure of that.

straussmaximilian, 2018-04-02:
From the resources I read the following:
`As a calibration reference, it uses two images (or stacks) of subdiffraction spots acquired in both channels to map the distortion.`
I guess what's usually used is a two-color bead sample? If the calibration is done with isolated spots, I potentially could implement such a function rather quickly with the 'Pick function.' Do you happen to have a sample dataset?

cleterrier, 2018-04-03:
I'm not sure what the Pick function refers to. Regarding a sample dataset, I will find a calibration pair (field of beads) and a two-color PAINT acquisition for testing.

The idea is to use a field of beads that are fluorescent in the different channels you image. Then you have to choose a function to make those two fields register at the nanoscale level: fit where are the bead center (using Tetraspeck beads you get down to a few nm error), and find the find the transform that make the center of each bead colocalize in the two (or more) channels. 

The question is what transform to use, a simple translation or affine transform is not enough at this scale, so you can use a secondary polynomial transform (see attached article) or B-splines (as done for DoM https://github.com/ekatrukha/DoM_Utrecht/wiki/Make-chromatic-calibration).

[2013_Opt. Express_Erdelyi.pdf](https://github.com/jungmannlab/picasso/files/1870665/2013_Opt.Express_Erdelyi.pdf)

cleterrier, 2018-04-03:
Here are two image stacks that can be used for calculation the transform: 4 corresponding bead field images with the far red (647) or red (561) lasers.
[Chromatic.zip](https://github.com/jungmannlab/picasso/files/1873154/Chromatic.zip)

You can max project two or more of them to have more beads (so more determination points for the transform)

straussmaximilian, 2018-04-05:
I did check the image stacks which resulted in the following vector map.

[image]
[image]

For some reason, there is an increase in deviation from left to right and up and down, where as I would expect a center position from where the deviation increases. I don't think its alignment as this would be just a constant offset.

Is this distribution something you would expect or do you have any idea what could cause this.

Any chance to have an image stack with more data points?

cleterrier, 2018-04-05:
Nice! This is done with the same filter cube and optical path, just changing the laser line. This is the center quadrant (256x256) of a 512x512 EMCCD, so it's really the core center of the field of view, where radial chromatic aberrations should be well compensated. At that scale (0.1 nm would be 16 nm), well below the specs of objective achromatic correction, I don't expect to find a theoretically explainable result. That is why people fit a polynomial or B-spline curve, because the deformation field is rarely simple (although it is spatially continuous and varies quite slowly, hence the use of polynomial/B-spline).