jungmannlab/picasso #18: Wishlist: Fixed ROIs for 3D calibration

hiroshisasak1, 2018-04-23:
For 3D calibration, Picasso Localize identifies and localizes fluorescent beads in the FoV frame by frame in the same way as localization for SMLM data. I recently calibrated beads data with astigmatism and realized that every fitting ROI was vertically shifted from the actual center of the signal when Picasso Localize analyzed the frames above the focal plane (see the attached screenshot). Due to this shift, Z localization precision above the focal plane is significantly worse than that below the focal plane.

It would be great if Picasso Localize uses fixed ROIs for 3D calibration, e.g. presets the fitting ROIs based on the maximum Z projection and doesn't move them frame by frame. Alternatively, if user can define ROIs, that would also work.

[image]

straussmaximilian, 2018-04-23:
I also noticed that the ROIs wiggle around during z-calibration but didn't see that they move in a particular direction. For this random wiggling, I would not assume that this unidirectionally changes the localization precision. The reason for doing the identification framewise is that the beads could move over time and the x and y position is not part of the calibration. For the TIRF calibration curves I saw I never noticed that the z localization is significantly worse above the focal plane than below. That is essentially the last plot in the picasso calibration figure (see attached image for reference)

Nevertheless, if you choose 'load picks as identifcations', Picasso will ask you for a pick regions file and a possible drift file. This will give you ROIs at the position of the pick that will stay over time (unless you specified a drift file). Be sure to set the correct ROI side beforehand.

I just made a change so that this feature can be used for the z-calibration. You can get it from the dev branch (https://github.com/jungmannlab/picasso/tree/dev) or drop me an E-mail if you want a compiled installer.

If this indeed improves the localization precision let me know, I am interested.
[image]

hiroshisasak1, 2018-04-23:
Thank you, Maximilian! I've just installed the updated version of Picasso to try out the 'load picks as identifications' function but faced an error as attached. In addition, I am wondering how a pick region file should look like. Is the same format for a Picasso Render pick region file (Centers + Diameter) good?
[image]

For you information, see the attached image of stage position vs. mean z localization plot. I will definitely let you know how it goes with fixed ROIs.
[image]

straussmaximilian, 2018-04-23:
Did you checkout to the dev branch? Try: `git checkout dev`. I noticed the error as well so I quickly fixed it. But it's not on master just on dev. As for the picked regions file: It is exactly the same format, so Ideally you reconstruct the calibration file, pick the regions in render where you see localizations and then load it into localize again. Note that the diameter will not be taken into account, so be sure to set the correct box size before loading. The plot indeed looks like that you have worse z precision below the focal plane. Is there anything in your optical path that could cause this?

hiroshisasak1, 2018-04-23:
I did 3D calibration from the same data but with fixed ROIs and it now looks much better than before, though I could improve it further. 
[image]

Just for your reference, the stage position (=z height) is flipped in the previous plot, so I had worse precision below the focal plane. We installed an adaptive optics system into our microscope, so PSF looks almost ideal. Therefore, I think the z-precision issue I had was due to ROI shift during localization.

straussmaximilian, 2018-04-23:
That looks great. The part with the adaptive optics makes sense to me. I'm happy if it works now.

closed 2018-10-25 by straussmaximilian