jungmannlab/picasso #19: Mac Os compability

straussmaximilian, 2018-04-24:
I noted that some of the functions don't work properly when using Picasso on Mac OS X. I'll use this thread to track all the issues.

- Show traces shows an Attribute Error: AttributeError: 'FigureCanvasMac' object has no attribute ‘size'
- Fitting with MLE is more error-prone on MacOS than for Windows with the same localization settings.

chaosunlucas, 2018-10-21:
> I noted that some of the functions don't work properly when using Picasso on Mac OS X or Linux. I'll use this thread to track all the issues.
> 
> Mac Os X:
> 
> * Show traces shows an Attribute Error: AttributeError: 'FigureCanvasMac' object has no attribute ‘size'
> * Fitting with MLE is more error-prone on MacOS than for Windows with the same localization settings.
> 
> Linux:
> 
> * Simulate does not add the *.raw ending on simulated files automatically.
> * Localize does not find any *.raw and *.tif files in the directory although they are there.

'Error prone' during MLE fitting on Mac means stuck during fitting? I have been having this problem doing MLE fitting on Mac. It would appear to freeze at some random point of fitting while the computer ram is far from fully used. Any solution to this yet?

straussmaximilian, 2018-10-21:
Hi,
Yes, exactly. It randomly gets stuck. The odd thing is that it is really platform dependent, i.e., the same dataset will go through on Windows but not on Mac. So far I did not have any luck figuring out what the problem is (or spotting an error message), so if you have any suggestions, let me know. Performance wise the MLE is a bit better than the LQ, though. Please find attached the comparison of MLE <> LQ for a simulated dataset with known ground truth.

[image]

chaosunlucas, 2018-10-22:
> Hi,
> Yes, exactly. It randomly gets stuck. The odd thing is that it is really platform dependent, i.e., the same dataset will go through on Windows but not on Mac. So far I did not have any luck [...]

Today I also had this problem for Windows. It occurred during a 14 million spot fitting with MLE. 
I also tried running LQ on Mac several times and every time python crashes at the beginning of fitting...

straussmaximilian, 2018-10-22:
Hi, could you elaborate a bit on the crashes?
In the past, I have seen that the localizer can crash under rare circumstances when selecting a low gradient and hence trying to fit localizations with low SNR in Windows. This typically meant always crashing at the same localization and could be solved with increasing the gradient. In contrast to that in OS X, it seemed random. Do you get any error message for the LQ crash on Mac Os?

chaosunlucas, 2018-10-23:
I will get back to you on Mac OS crashing errors. I have another question about Windows. If I were to purchase a PC and use windows, is there any compatibility issue? For example, is Windows 10 OK? Does it require specific processors?  Thanks!

chaosunlucas, 2018-10-23:
> Hi, could you elaborate a bit on the crashes?
> In the past, I have seen that the localizer can crash under rare circumstances when selecting a low gradient and hence trying to fit localizations with [...]

See above

straussmaximilian, 2018-10-23:
There are no Windows compatibility issues that I am aware of. So far I had it working on Win8 / 10, Server 2010 and 2012. For GPU functionality you will need an NVIDIA graphics card, though. Depending on how many users should be working simultaneously on your machine I would pick the specs accordingly. 
If you drop me an email, I could send you the specifications of our workstation.

chaosunlucas, 2018-10-23:
> There are no Windows compatibility issues that I am aware of. So far I had it working on Win8 / 10, Server 2010 and 2012. For GPU functionality you will need an NVIDIA graphics card, though. [...]

Thanks! I will email you right away. Meanwhile, there was no error message for LQ crash on Mac OS. I got a message as below:
[image]

and a error report:
[Process error.docx](https://github.com/jungmannlab/picasso/files/2506167/Process.error.docx)