jungmannlab/picasso #20: Problem with 3D calibration

aaristov, 2018-06-18:
I try to calibrate astigmatism from the stack https://www.dropbox.com/s/q565b0bcjyz1f12/cal-ast-5nm_1-crop.tif?dl=0

And polynomial fit looks weird:
https://www.dropbox.com/s/emo8my51tund3ud/3d_cal.png?dl=0

I set the following parameters:
Box size 16
min Net Grad 810
EM gain 1 (data in photon counting mode)
Baseline 0
pixel size 110
Method: LQ
Step size: 5 nm

I also copy the terminal output:

> C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\core\_methods.py:
> 59: RuntimeWarning: Mean of empty slice.
>  warnings.warn("Mean of empty slice.", RuntimeWarning)
> C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\core\_methods.py:
> 70: RuntimeWarning: invalid value encountered in true_divide
>  ret = ret.dtype.type(ret / rcount)

> Intel MKL ERROR: Parameter 6 was incorrect on entry to DGELSD.
> C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\lib\polynomial.py
> :595: RankWarning: Polyfit may be poorly conditioned
>   warnings.warn(msg, RankWarning)

> Intel MKL ERROR: Parameter 6 was incorrect on entry to DGELSD.
> C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\lib\polynomial.py
> :595: RankWarning: Polyfit may be poorly conditioned
>  warnings.warn(msg, RankWarning)
> C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\core\_methods.py:
>59: RuntimeWarning: Mean of empty slice.
>  warnings.warn("Mean of empty slice.", RuntimeWarning)
>C:\Users\Andrey\Anaconda3\envs\picasso\lib\site-packages\numpy\core\_methods.py:
>70: RuntimeWarning: invalid value encountered in double_scalars
>  ret = ret.dtype.type(ret / rcount)
>QCoreApplication::exec: The event loop is already running

straussmaximilian, 2018-06-18:
Hi, I could reproduce the problem. Apparently, the fit is not converging, and this is causing problems.
I noticed that your PSFs are quite large, so I did a sum-binning in ImageJ and was able to get a better polynomial fit:

[image]

I would recommend that you activate binning on your camera. Additionally, you could increase the z-range (from, e.g. -500 to +500). Note that Picasso always assumes that you are going from minus to plus.

aaristov, 2018-06-18:
There is localization filter in the code which removes some localizations with low certainty, but z_range vector stays unaffected. That's why polyfit doesn't work. Pull request sent.

straussmaximilian, 2018-06-18:
Hi,
I checked this again and noticed the nans. I guess we never experienced that as we usually have more spots and higher SNR. I now made an additional check that interpolates the nans, and the fit worked smoothly. Feel free to check out #995 in the dev branch.

[image]

closed 2018-08-16 by aaristov