jungmannlab/picasso #23: 3D reconstruction parameters log

aaristov, 2018-08-16:
I found lots of parameters in the yaml file after localizing astigmatism, but one important is missing: Which calibration has been used. I would like to have it in the yaml file along with the camera gain-baseline settings.
Thanks.
[image]

straussmaximilian, 2018-08-17:
I understand, that would be useful. Would you rather have the file path of the configuration file that has been used or the fit coefficients?

aaristov, 2018-08-17:
Exactly. Would be great also to load automatically calibration file after saving it.

cleterrier, 2018-08-17:
So I'd say both: file path and fit coefficients

straussmaximilian, 2018-08-20:
Commit #1023 in the dev branch. 
[image]

aaristov, 2018-08-21:
Fantastic! Thanks a lot.

closed 2018-08-21 by aaristov