jungmannlab/picasso #24: RCC drift correction fails when installing via Python

Fnunesv, 2018-09-22:
Hey

I had a version of your software provided by your group to our PI (Gregory Giannone) which I've been using so far with no issues. Recently I've decided to re-install it via Python to have access to other routines such as the .csv exporter in the Render file. However, after this, RCC drift correction stopped working. Before that I never had any issues. Now, when using the same .hdf5 file with our standard version or the Python-installed one, the latter always gives me an error message when performing RCC: "RCC failed. Consider changing segmentation and make sure there are enough locs per frame"

It's not dependent on segmentation as I have increased it up to 100000 and still get the same error message. Any idea why this might be happening? I updated the tqdm package using anaconda, but no changes. Below you can see a comparison of both cases, with RCC working in the standard version and the error message for the Python version. 

[image]

[image]

Fnunesv, 2018-09-24:
Hey!

Just to let you know the problem is solved. I launched Render directly from the Anaconda terminal instead of double-clicking the shortcut (python -m picasso render). RCC drift correction now works. If any of the other users have this issue, launching directly from command line should solve it.

straussmaximilian, 2018-09-24:
Hi,
I am glad that things worked out. At some point, I added this RCC error message whenever calling RCC yields an exception. However, I didn't change the RCC function itself. 
Anyhow, I am aware of the mess that we have serval one-click-installers with unknown versions floating around, so tracking this bug might be a bit tricky. In the future, I am planning on implementing a better versioning system. Nevertheless, if you want, you can drop me an e-mail, and I could provide you with the latest installer that I compiled.

Fnunesv, 2018-09-26:
Hey,

Yes, but thanks for the reply anyway! I will for sure.

closed 2018-10-12 by straussmaximilian