jungmannlab/picasso #29: Localize Freezes while fitting the spots

AmayAgrawal, 2018-11-28:
I have a raw data containing 1,00,000 frames but after opening it in localize, selecting the net gradient and allowing it to "Identify and fit", it identifies spots in all the frames and starts fitting but while fitting after a certain time interval the fitting freezes at one point and throws the memory error. Then, I tried to reduce the frames to 50,000 and again started fitting, this time it is not showing memory error but it still freezes after fitting some points. Also, I tried increasing the net gradient and again started the fitting, but still no luck!

straussmaximilian, 2018-11-28:
Hi,
Thanks for reporting this issue. Which fitting method did you use? Does this error also occur when using LQ? Did you correctly set your camera parameters? Do you mind uploading and sharing the 50k subset so I could try and troubleshoot?

AmayAgrawal, 2018-11-29:
Hi,
For fitting method I can't see any option for selecting particular method for fitting. I am attaching the screenshot here.
[image]

For Camera parameters I have set it correctly. I am uploading the zip folder containing the 50k subset file, so you can try and troubleshoot.

Here is the link for the file : [https://drive.google.com/open?id=1DUUOMYwNQdGd4sC4jkYGTYbAoJ_uPaLO](url)

**P.S** : Can you please let me know as soon as you download the above folder, so that I can remove the folder from my drive as I have limited space on drive

straussmaximilian, 2018-11-29:
Hi, do you have the latest Picasso version installed? That's 0.2.2, and the installer can be found on https://github.com/jungmannlab/picasso/releases. This should have the dropdown for the fitting methods.
I downloaded the file. As it is a raw file could you also provide me with the byte Order, the width and height? I assume it's 50.000 frames, right?
As for the camera parameters, do you use the values from the above screenshot?
If yes do you mind double-checking? A QE of 1.0 seems very unlikely to me. Usually, your camera has a spec sheet where those values are listed.

AmayAgrawal, 2018-11-29:
Hi,
I have installed the latest version. The Byte order is : Big Endian, Data type : int16, Frames : 50,000, height and width : 256, EM gain : 70, Baseline : 0, Quantum Efficiency : 0.95. 

I am experiencing this freezing problem with not only this file but in the past I have experienced it with other files also. Also, I will again try to fit it with "LQ Gaussian Method" and let you if I am able to do it or not.

straussmaximilian, 2018-11-29:
Hi,
so I tried it with MLE and it froze at localization 3243123. There where no problems with LQ and I could reconstruct the file. Let me know how the LQ worked out for you!
[image]

straussmaximilian, 2018-12-06:
Hi, I was able to dedicate some time today for troubleshooting. The MLE bug should now be fixed with v0.2.4 (release page). Feel free to check it out and let me know in case the bug persists.

closed 2018-12-06 by straussmaximilian