jungmannlab/picasso #30: Render crashes with exception

cleterrier, 2018-12-09:
Just tried the latest version on OSX (installed from source). Localize and Filter work OK so far. Render crashes with:

`
(picasso) MacBook-Pro-5:~ christo$ python -m picasso render
/Users/christo/anaconda3/envs/picasso/lib/python3.5/site-packages/sklearn/utils/fixes.py:313: FutureWarning: numpy not_equal will not check object identity in the future. The comparison did not return the same result as suggested by the identity (`is`)) and will change.
  _nan_object_mask = _nan_object_array != _nan_object_array
2018-12-09 15:43:54.033 python[17445:1342641] -[QNSApplication _setup:]: unrecognized selector sent to instance 0x7fe4b9c26c70
2018-12-09 15:43:54.035 python[17445:1342641] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[QNSApplication _setup:]: unrecognized selector sent to instance 0x7fe4b9c26c70'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff3a634e65 __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff6668f720 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3a6b222d -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007fff3a5d6820 ___forwarding___ + 1486
	4   CoreFoundation                      0x00007fff3a5d61c8 _CF_forwarding_prep_0 + 120
	5   libtk8.6.dylib                      0x000000010f92a475 TkpInit + 310
	6   libtk8.6.dylib                      0x000000010f89a754 Initialize + 2457
	7   _tkinter.cpython-35m-darwin.so      0x000000010f6fcd64 Tcl_AppInit + 84
	8   _tkinter.cpython-35m-darwin.so      0x000000010f6fca3b _tkinter_create + 1115
	9   libpython3.5m.dylib                 0x000000010448776c PyCFunction_Call + 220
	10  libpython3.5m.dylib                 0x000000010450458b PyEval_EvalFrameEx + 25515
	11  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	12  libpython3.5m.dylib                 0x00000001044fe1be PyEval_EvalCodeEx + 78
	13  libpython3.5m.dylib                 0x0000000104467ffd function_call + 381
	14  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	15  libpython3.5m.dylib                 0x000000010445450d method_call + 141
	16  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	17  libpython3.5m.dylib                 0x000000010449e3bd slot_tp_init + 125
	18  libpython3.5m.dylib                 0x000000010449a9d9 type_call + 297
	19  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	20  libpython3.5m.dylib                 0x00000001045042d0 PyEval_EvalFrameEx + 24816
	21  libpython3.5m.dylib                 0x00000001045096b0 fast_function + 304
	22  libpython3.5m.dylib                 0x000000010450400b PyEval_EvalFrameEx + 24107
	23  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	24  libpython3.5m.dylib                 0x00000001044fe1be PyEval_EvalCodeEx + 78
	25  libpython3.5m.dylib                 0x0000000104467ffd function_call + 381
	26  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	27  libpython3.5m.dylib                 0x0000000104504dde PyEval_EvalFrameEx + 27646
	28  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	29  libpython3.5m.dylib                 0x0000000104509731 fast_function + 433
	30  libpython3.5m.dylib                 0x000000010450400b PyEval_EvalFrameEx + 24107
	31  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	32  libpython3.5m.dylib                 0x00000001044fe1be PyEval_EvalCodeEx + 78
	33  libpython3.5m.dylib                 0x0000000104467ffd function_call + 381
	34  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	35  libpython3.5m.dylib                 0x000000010445450d method_call + 141
	36  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	37  libpython3.5m.dylib                 0x000000010449e3bd slot_tp_init + 125
	38  libpython3.5m.dylib                 0x000000010449a9d9 type_call + 297
	39  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	40  libpython3.5m.dylib                 0x00000001045042d0 PyEval_EvalFrameEx + 24816
	41  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	42  libpython3.5m.dylib                 0x00000001044fe1be PyEval_EvalCodeEx + 78
	43  libpython3.5m.dylib                 0x0000000104467ffd function_call + 381
	44  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	45  libpython3.5m.dylib                 0x000000010445450d method_call + 141
	46  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	47  libpython3.5m.dylib                 0x000000010449e3bd slot_tp_init + 125
	48  libpython3.5m.dylib                 0x000000010449a9d9 type_call + 297
	49  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	50  libpython3.5m.dylib                 0x00000001045042d0 PyEval_EvalFrameEx + 24816
	51  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	52  libpython3.5m.dylib                 0x0000000104509731 fast_function + 433
	53  libpython3.5m.dylib                 0x000000010450400b PyEval_EvalFrameEx + 24107
	54  libpython3.5m.dylib                 0x00000001045096b0 fast_function + 304
	55  libpython3.5m.dylib                 0x000000010450400b PyEval_EvalFrameEx + 24107
	56  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	57  libpython3.5m.dylib                 0x00000001044fe161 PyEval_EvalCode + 81
	58  libpython3.5m.dylib                 0x00000001044fb7a7 builtin_exec + 551
	59  libpython3.5m.dylib                 0x000000010448776c PyCFunction_Call + 220
	60  libpython3.5m.dylib                 0x000000010450458b PyEval_EvalFrameEx + 25515
	61  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	62  libpython3.5m.dylib                 0x0000000104509731 fast_function + 433
	63  libpython3.5m.dylib                 0x000000010450400b PyEval_EvalFrameEx + 24107
	64  libpython3.5m.dylib                 0x0000000104508ef9 _PyEval_EvalCodeWithName + 3945
	65  libpython3.5m.dylib                 0x00000001044fe1be PyEval_EvalCodeEx + 78
	66  libpython3.5m.dylib                 0x0000000104467ffd function_call + 381
	67  libpython3.5m.dylib                 0x000000010443e491 PyObject_Call + 97
	68  libpython3.5m.dylib                 0x0000000104547533 RunModule + 147
	69  libpython3.5m.dylib                 0x0000000104546f6e Py_Main + 2686
	70  python                              0x0000000104427e18 main + 216
	71  libdyld.dylib                       0x00007fff6775e08d start + 1
	72  ???                                 0x0000000000000004 0x0 + 4
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
`

cleterrier, 2018-12-09:
Just in case, I re-did the whole install process on a new conda environment, using a downloaded 0.2.4 release. Here's my setup:

Last login: Sun Dec  9 16:53:58 on ttys000
MacBook-Pro-5:~ christo$ source activate picasso
(picasso) MacBook-Pro-5:~ christo$ conda list
# packages in environment at /Users/christo/anaconda3/envs/picasso:
#
# Name                    Version                   Build  Channel
asteval                   0.9.13                    <pip>
blas                      1.1                    openblas    conda-forge
bzip2                     1.0.6                         1    conda-forge
ca-certificates           2018.11.29           ha4d7672_0    conda-forge
certifi                   2018.8.24             py35_1001    conda-forge
colorama                  0.4.0                      py_0    conda-forge
cycler                    0.10.0                     py_1    conda-forge
freetype                  2.6.3                         1    conda-forge
h5py                      2.8.0            py35h7eb728f_2    conda-forge
hdf5                      1.10.2               hc401514_3    conda-forge
libffi                    3.2.1                hfc679d8_5    conda-forge
libgfortran               3.0.0                         1    conda-forge
libpng                    1.6.36               ha92aebf_0    conda-forge
llvmlite                  0.25.0                   py35_0    conda-forge
lmfit                     0.9.12                    <pip>
matplotlib                1.5.1               np111py35_3    conda-forge
ncurses                   6.1                  hfc679d8_1    conda-forge
numba                     0.40.0           py35hf8a1672_0    conda-forge
numpy                     1.11.3          py35_blas_openblashd3ea46f_205  [blas_openblas]  conda-forge
openblas                  0.2.20                        8    conda-forge
openssl                   1.0.2p               h470a237_1    conda-forge
picasso                   0.2.4                     <pip>
pip                       18.0                  py35_1001    conda-forge
pyparsing                 2.3.0                      py_0    conda-forge
pyqt                      4.11.4                   py35_2    conda-forge
python                    3.5.5                h5001a0f_2    conda-forge
python-dateutil           2.7.5                      py_0    conda-forge
pytz                      2018.7                     py_0    conda-forge
pyyaml                    3.13             py35h470a237_1    conda-forge
qt                        4.8.7                         4  
readline                  7.0                  haf1bffa_1    conda-forge
scikit-learn              0.20.0          py35_blas_openblasha84fab4_201  [blas_openblas]  conda-forge
scipy                     1.1.0           py35_blas_openblash7943236_201  [blas_openblas]  conda-forge
setuptools                40.4.3                   py35_0    conda-forge
sip                       4.18                     py35_1    conda-forge
six                       1.11.0                   py35_1    conda-forge
sqlite                    3.26.0               hb1c47c0_0    conda-forge
tk                        8.6.9                ha92aebf_0    conda-forge
tqdm                      4.28.1                     py_0    conda-forge
uncertainties             3.0.3                     <pip>
wheel                     0.32.0                py35_1000    conda-forge
xz                        5.2.4                h470a237_1    conda-forge
yaml                      0.1.7                h470a237_1    conda-forge
zlib                      1.2.11               h470a237_3    conda-forge

cleterrier, 2018-12-10:
do you think a difference in the environment can cause the crash I'm seeing? I exactly followed the instructions on the readme for OSX install from source but end up with a few elements that are different.

cleterrier, 2018-12-10:
Here is the full log from creating the environment to the crash. What am I doing wrong?

Last login: Mon Dec 10 11:29:41 on console
MacBook-Pro-5:~ christo$ conda create --name picasso python=3.5
Solving environment: done

## Package Plan ##

  environment location: /Users/christo/anaconda3/envs/picasso

  added / updated specs: 
    - python=3.5

The following NEW packages will be INSTALLED:

    bzip2:           1.0.6-1               conda-forge
    ca-certificates: 2018.11.29-ha4d7672_0 conda-forge
    certifi:         2018.8.24-py35_1001   conda-forge
    libffi:          3.2.1-hfc679d8_5      conda-forge
    ncurses:         6.1-hfc679d8_1        conda-forge
    openssl:         1.0.2p-h470a237_1     conda-forge
    pip:             18.0-py35_1001        conda-forge
    python:          3.5.5-h5001a0f_2      conda-forge
    readline:        7.0-haf1bffa_1        conda-forge
    setuptools:      40.4.3-py35_0         conda-forge
    sqlite:          3.26.0-hb1c47c0_0     conda-forge
    tk:              8.6.9-ha92aebf_0      conda-forge
    wheel:           0.32.0-py35_1000      conda-forge
    xz:              5.2.4-h470a237_1      conda-forge
    zlib:            1.2.11-h470a237_3     conda-forge

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use:
# > source activate picasso
#
# To deactivate an active environment, use:
# > source deactivate
#

MacBook-Pro-5:~ christo$ source activate picasso
(picasso) MacBook-Pro-5:~ christo$ conda install h5py matplotlib=1.5.1 numba numpy scipy pyqt=4 pyyaml scikit-learn colorama tqdm
Solving environment: done

## Package Plan ##

  environment location: /Users/christo/anaconda3/envs/picasso

  added / updated specs: 
    - colorama
    - h5py
    - matplotlib=1.5.1
    - numba
    - numpy
    - pyqt=4
    - pyyaml
    - scikit-learn
    - scipy
    - tqdm

The following NEW packages will be INSTALLED:

    blas:            1.1-openblas                          conda-forge
    colorama:        0.4.0-py_0                            conda-forge
    cycler:          0.10.0-py_1                           conda-forge
    freetype:        2.6.3-1                               conda-forge
[... 261 lines cut ...]
	51  libpython3.5m.dylib                 0x000000010676cef9 _PyEval_EvalCodeWithName + 3945
	52  libpython3.5m.dylib                 0x000000010676d731 fast_function + 433
	53  libpython3.5m.dylib                 0x000000010676800b PyEval_EvalFrameEx + 24107
	54  libpython3.5m.dylib                 0x000000010676d6b0 fast_function + 304
	55  libpython3.5m.dylib                 0x000000010676800b PyEval_EvalFrameEx + 24107
	56  libpython3.5m.dylib                 0x000000010676cef9 _PyEval_EvalCodeWithName + 3945
	57  libpython3.5m.dylib                 0x0000000106762161 PyEval_EvalCode + 81
	58  libpython3.5m.dylib                 0x000000010675f7a7 builtin_exec + 551
	59  libpython3.5m.dylib                 0x00000001066eb76c PyCFunction_Call + 220
	60  libpython3.5m.dylib                 0x000000010676858b PyEval_EvalFrameEx + 25515
	61  libpython3.5m.dylib                 0x000000010676cef9 _PyEval_EvalCodeWithName + 3945
	62  libpython3.5m.dylib                 0x000000010676d731 fast_function + 433
	63  libpython3.5m.dylib                 0x000000010676800b PyEval_EvalFrameEx + 24107
	64  libpython3.5m.dylib                 0x000000010676cef9 _PyEval_EvalCodeWithName + 3945
	65  libpython3.5m.dylib                 0x00000001067621be PyEval_EvalCodeEx + 78
	66  libpython3.5m.dylib                 0x00000001066cbffd function_call + 381
	67  libpython3.5m.dylib                 0x00000001066a2491 PyObject_Call + 97
	68  libpython3.5m.dylib                 0x00000001067ab533 RunModule + 147
	69  libpython3.5m.dylib                 0x00000001067aaf6e Py_Main + 2686
	70  python                              0x0000000106689e18 main + 216
	71  libdyld.dylib                       0x00007fff73c41ed9 start + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
(picasso) MacBook-Pro-5:picasso christo$

cleterrier, 2018-12-10:
tried to create the environment directly from the anaconda GUI, tried installing or not the pyinstaller... I always get the same result. FWIW I'm under OSX 10.14.2.

cleterrier, 2018-12-10:
completely reinstalled anaconda, redid the whole install... same thing

cleterrier, 2018-12-10:
resetting conda priorities (as explained here https://stackoverflow.com/questions/48547046/resetting-conda-channel-priorities) back to default rather than conda-forge and reinstalling a new environment and Picasso seem to have solved the issue. Maybe add a warning about what package channel to use?

straussmaximilian, 2018-12-11:
Hi, great that you got it to work. I will add a warning about the package channel.

closed 2018-12-11 by straussmaximilian