jungmannlab/picasso #32: RCC drift correction failed

hiroshisasak1, 2018-12-18:
This is just a bug report related to #24 and #27.

I had exactly the same issue and found that the failure is related to pythonw. When I launch Render via pythonw.exe (pythonw.exe -m picasso render), RCC always crashes regardless of whether it is launched from command line or shortcut. Launching Render via python.exe (python.exe -m picasso render) solves this.

straussmaximilian, 2018-12-19:
Hi,
Thanks for sharing. When launching via command line, is there any error message displayed?

hiroshisasak1, 2018-12-20:
No comments appear as it happens only when I launch Render using pythonw. For your reference, here's the detail of the error I encounter:

1. Launch Render via pythonw (pythonw -m picasso render)
2. Open a localization file that doesn't have much drift
3. Select "Undrift by RCC"
4. Enter a segmentation value (No matter what I enter, the following error occurs)
5. Press OK
6. The RCC Error message box shows up saying "RCC failed. Consider changing segmentation and make sure there are enough locs per frame"
7. Half a second after 6, the "Generating segments" box appears remaining 0%
8. Render crashes, have to kill the process through Task Manager

I will install Picasso on Mac and check out this later.

straussmaximilian, 2019-02-20:
I now changed the error handling for RCC to display what's causing the crash. This way we can troubleshoot what happened. I am closing this for now.

closed 2019-02-20 by straussmaximilian