jungmannlab/picasso #34: Fix version pinning between Redme and requirements.txt

mpibpc-mroose, 2019-01-29:
Hi all,
in the ReadMe there is `matplotlib==1.5.1` mentioned as requirements, but in your `requirements.txt` no version pinning occurs. So this has to be fixed.

Additionally the PyQt4 requirement is missing in `requirements.txt`.

aaristov, 2019-01-29:
PyQt4 and matplotlib are not available via pypi, so you have to install it via conda install.

straussmaximilian, 2019-01-31:
Hi, thanks for pointing this out. The version pinning and the additional install via conda while having a requriemtens.txt is indeed not satisfactory. 

In the future, there will be a bigger update from PyQt4 to PyQt5 (which is installable via pip) plus I'll fix the compatibility issues for matplotlib so that a complete install via requirements.txt should be possible.

straussmaximilian, 2020-01-06:
Version pinning now maintained with dependabot.

closed 2020-01-06 by straussmaximilian