jungmannlab/picasso #35: How to use GPUfit with the one-click installer?

cleterrier, 2019-02-02:
Hi, I just installed 0.2.5 via the one-click installer on Windows 10, works like a charm. I was wondering how I can add GPUfit capability to the install, as I could not find this in the docs. Thanks!

Mark-Dfly, 2022-04-26:
Hi Max,

Did this ever get resolved? Is a GPU fit version available?

straussmaximilian, 2022-04-26:
Hi Mark,
I never followed up on this and until now the GPU fit version only works with the Python-build.
However I now thought about this again and I think there is a potential way to include GPU-Fit in the one-click build.

I just made a small test - seems I can include the DLL. However this required to downgrade from Python3.8 to Python3.7 for GPUFit and this could lead to compatibility issues with the current Picasso-build. Would therefore require an update for the Gpufit library to work w/o issues.

Will give another shot and then can estimate this better.

cleterrier, 2023-07-26:
Just checking what is the current state of this? Due to problems in building from source, I'm using the one-click installer. I tried to add the current GPUfit python packages (pygpufit and pyGpufit-1.2.0.dist-info) in the Picasso install (C:\Program Files\Picasso), but I end up with this error:

```
Traceback (most recent call last):
  File "PyInstaller\loader\pyimod04_ctypes.py", line 53, in __init__
  File "ctypes\__init__.py", line 373, in __init__
FileNotFoundError: Could not find module 'C:\Program Files\Picasso\pygpufit\Gpufit.dll' (or one of its dependencies). Try using the full path with constructor syntax.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "picasso_pyinstaller.py", line 9, in <module>
  File "picasso\__main__.py", line 1658, in main
    from picasso.gui import localize
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "picasso\gui\localize.py", line 20, in <module>
    from .. import io, localize, gausslq, gaussmle, zfit, lib, CONFIG, avgroi
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "PyInstaller\loader\pyimod03_importers.py", line 495, in exec_module
  File "picasso\gausslq.py", line 21, in <module>
    from pygpufit import gpufit as gf
  File "C:\Program Files\Picasso\pygpufit\gpufit.py", line 24, in <module>
    lib = cdll.LoadLibrary(lib_path)
  File "ctypes\__init__.py", line 451, in LoadLibrary
  File "PyInstaller\loader\pyimod04_ctypes.py", line 55, in __init__
pyimod04_ctypes.install.<locals>.PyInstallerImportError: Failed to load dynlib/dll 'C:\\Program Files\\Picasso\\pygpufit\\Gpufit.dll'. Most likely this dynlib/dll was not found when the application was frozen.
```

straussmaximilian, 2023-08-01:
Hi,
I stopped working on this, maybe @rafalkowalewski1 wants to investigate this?
I think in principle this should work.

Some additional information:
- We have a new way to compile the installer that should be better at getting the imports. This is the issue here as simply copying the file doesn't do the trick
- Including In the requirements is not straightforward as pygpufit is not at PyPi
- However, you can also include wheels in the requirements.txt via direct references in the newer versions. This could do the trick (see https://peps.python.org/pep-0440/#direct-references)