jungmannlab/picasso #36: Error closing dataset in render

aaristov, 2019-02-12:
I've loaded two datasets in Render and wanted to close one. I opened Files, clicked X and got this error 
[image]

Also, this is my terminal output:

> 
> 2019-02-12 17:59:02.659 python[78978:39778850] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
> Error in sys.excepthook:
> Traceback (most recent call last):
>   File "/Users/andrey/anaconda3/envs/picasso/lib/python3.5/site-packages/picasso/gui/render.py", line 6448, in excepthook
>     errorbox.exec_()
> AttributeError: 'StandardButton' object has no attribute 'exec_'
> 
> Original exception was:
> Traceback (most recent call last):
>   File "/Users/andrey/anaconda3/envs/picasso/lib/python3.5/site-packages/picasso/gui/render.py", line 342, in close_file
>     print(self.sender.objectName())
> AttributeError: 'builtin_function_or_method' object has no attribute 'objectName'
> 

I'm using picasso-0.2.5 from master branch on Mac.

Thanks

straussmaximilian, 2019-02-12:
Thanks for letting me know. This is part of the new framework for file handling, but I haven't added the functionality yet. I'll keep this open so that others know.

straussmaximilian, 2019-02-14:
Added a NotImplementedError for now.

closed 2019-02-14 by straussmaximilian