jungmannlab/picasso #39: Batch processing with command line

kfriedl, 2019-02-18:
Hi, I am trying to set up a batch file to process data via command line. Unfortunately our data (*.raw) is not recognized by the module localize. Maybe glob is causing the error, a small test on the same directory with os.listdir() and fnmatch worked well.

straussmaximilian, 2019-02-18:
Hi,
when using raw files, you need to have a corresponding *.yaml file so that the image dimensions are known. 
I just pushed a commit that checks if each *.raw file has a *.yaml file. If not, the program asks for the image specifications. Here you can enter the details and choose to either be prompted again for all the other *.raw files or use the same specs for all remaining files.
Do you mind testing this out?
Also, let me know if you had a different workflow in mind.

kfriedl, 2019-02-18:
I converted the files to raw with your raw/yaml plugin, so the files are all there. I tested the change, but still get a FileNotFoundError,

straussmaximilian, 2019-02-18:
Hi, could you paste the error message? Which OS are you using?

kfriedl, 2019-02-18:
Windows 10
[image]

hiroshisasak1, 2019-02-18:
Hi all,
I think it is because, in check_consecutive_tif(filepath) in __main__.py, only .tif files are searched while running Picasso Localize via CUI.

Line 643 in __main__.py
>         files = glob(filepath + "/*.tif")

I use the code changing .tif to .raw by myself.

kfriedl, 2019-02-18:
I see! In line 669 it's glob(filepath+"*.raw"), maybe the slash is missing

straussmaximilian, 2019-02-18:
Hi,
The check_consecutive is unrelated; this is only meant to find consecutive TIFs and treat them as one. This is intentionally TIF only. A couple lines later you can find  `paths = tif_files + glob(files + "*.raw")` to include the *.raw files as kfriedl noted.

It seems that you have the installer version plus the GitHub version running. I tested both and noticed that the batch mode gets an error when calling the installer version (see Screenshots) and works fine when calling the GitHub version.

Therefore I would recommend calling "python -m picasso localize ..." to call the GitHub version. I'll investigate what's wrong with the installer. 

[image]
[image]

kfriedl, 2019-02-18:
So I had a try and changed line 669 to glob(files+"/.raw") and the slash does the trick. Sorry to have caused so much trouble for this minor problem!

straussmaximilian, 2019-02-18:
Hi, no trouble here - glad if things worked out. I changed it accordingly and will compile a new installer (v0.2.7).

closed 2019-02-18 by straussmaximilian