jungmannlab/picasso #40: files made with raw_yaml exporter plugin are not loadable into Localize

mwdocter, 2019-02-27:
I want to load images made through Andor Solis (*.dat and sifx) into Picasso. I already changed them into avi through Matlab (to analyze with ThunderSTORM in ImageJ). In order to do FRET analysis I want to use Picasso. 

The (really long) route, making a raw file through Matlab and loading it into Picasso works, at least for 512x1024pixels, 100 frames

However, I would love to use the much faster raw_yaml exporter in ImageJ. All raw files generated through raw_yaml exporter (even the ones with only one frame of 512x1024 pixels) give me the following OSerror about not enough storage. 
[image]

Since my Matlab generated raw files are much larger, the file size cannot be the cause of the issue. So can it somehow be I am not using the exporter plugin correctly? 
Checking previous issues, #26 File loading error in Localize module #26, was solved by updating Picasso to the latest version. I downloaded Picasso and the raw_yaml plugin like two weeks ago, so also that should not be the issue. 
Any help is appreciated.

straussmaximilian, 2019-02-27:
Hi,
Can you quickly check whether the values in the *.yaml file are correct? I.e., the Frames/Height & Width? Do you mind uploading one of these files for me so that I can troubleshoot?

akurz-wue, 2019-02-27:
one way to shorten your processing steps might be to spool files in tiff format directly via the solis software

mwdocter, 2019-02-27:
I checked the values, width, height and frames are correct. Data type should be uint16, and byte order I did not change

[files exported through raw_yaml .zip](https://github.com/jungmannlab/picasso/files/2911208/files.exported.through.raw_yaml.zip)

mwdocter, 2019-02-27:
@ akurz-wue: I'm using the Zyla 4.2, and need to spool my data. I have no control over the spooling, automatically I receive .dat files and a .sifx, which is needed to read/reconstruct in the dat files afterwards.

akurz-wue, 2019-02-27:
ah right, I remember that! But Andor provides a sif reader for imagej, maybe with batch conversion?

mwdocter, 2019-02-27:
and if it helps, here is the raw file generated in Matlab (+code)
[raw file through Matlab.zip](https://github.com/jungmannlab/picasso/files/2911262/raw.file.through.Matlab.zip)

mwdocter, 2019-02-27:
I'll look into the sifreader for imageJ (I am using it in Matlab), thanks for the tip. I could have known it also existed for ImageJ, and it will definitely speed up my process. Then only to solve what I'm doing wrong with the raw_yaml plugin, and why these raw files give an error loading them into Localize.

straussmaximilian, 2019-02-27:
Hi,
I was able to reproduce your problem and couldn't load the spool raw. Apparently, there seems to be a problem in the conversion from ImageJ. I couldn't even open the raw in ImageJ properly. I only could open the file by setting dimensions of 1024x256 instead of 1024x512. This allows to open the file, but it is stretched. Alternatively, I had to change the bit depth.

[image]

Interestingly though, the problem doesn't seem to arise from the plugin. If I open your Matlab Raw file, convert it to a tif, open it with ImageJ again and convert it to raw again with the plugin localize has no problem. I, therefore, would suggest to do the following: Could you double-check the file settings when having the file open in ImageJ (i.e., bit depth). After all, ImageJ should be able to open the raw after conversion. Additionally, I noted that your naming seems to be off, i.e., instead of 512x1024x1 it should be 1024x512x1.

mwdocter, 2019-02-28:
You are absolutely right on the bit-depth. While making the avi's in Matlab I had to rescale the maximum to 1, and apparently then the avi is stored at uint8 instead of uint16. In ImageJ the avi file has bit depth 8. 

So, the conclusion is to avoid the avi substep, therefore I tried to load the sifx/dat from Andor direclty into ImageJ, and then use the raw_yaml Plugin. So this issue can be closed. 

A small note on sifx/dat to ImageJ to raw/yaml:
Drawback of the sifreader is it cannot read in as virtual stack (to be solved by Andor). But when trying to use the raw_yaml plugin, I cannot export the data directly through the yaml plugin. 
[image]
If I save it first to a new tiff files, with type reduced to 16 bit , and then use the raw_yaml plugin I can save to raw and actually open in localize.

straussmaximilian, 2019-02-28:
Great that you find a way to get it to work. I am closing this now.

closed 2019-02-28 by straussmaximilian