jungmannlab/picasso #41: found locations are pixelated

mwdocter, 2019-03-04:
When running Localize the spot locations are pixelized, meaning they appear at distinct grid points, and hardly in between. However, if I run the same file through ImageJ/Thunderstorm I do not get this issue. Please have a look at the images below.

Surely I must be doing something wrong in the settings of Picasso, but I cannot figure out what. The raw data is uint8, 101 by 101 pixels, 5000 frames. 

the data can be found on wetransfer (>10MB) https://we.tl/t-sTdPUHXCTQ 

[image]
[image]

straussmaximilian, 2019-03-04:
Did you select "Average" as fit method? This only returns the average intensity of the spot and does not determine the center position. Hence the pixelation. LQ/MLE should work just fine. 
[image]

mwdocter, 2019-03-04:
I did, also before (I double checked and even ran the other three settings as well). 
It seems like somehow the previous settings (I did try all fit methods previously) are stored in cache. 
Running it again after restarted Picasso, gives the image you show. 

However, I also managed by changing the methods to average, rerun Localize and then rerun again with MLE method, that the pixelation occurs again. Sorry for the vague description when this happens, I manage to get this once more today and definitely in all runs last week. It seems to me some setting somehow is not updated correctly, or remaining from a previous session. If I find how to reproduce, I will let you know.

straussmaximilian, 2019-03-04:
Okay. Can you let me know which Picasso version you are using? I.e. the one-click installer version?

mwdocter, 2019-03-04:
I'm using the one-click installer, version picasso-0.2.5

mwdocter, 2019-03-06:
I tried it again today: 
1. run Localize on a raw file, with default parameters. Then the rendered image is pixelated again. When checking the parameters the method is MLE.
2. played around with the parameters. When setting the EMgain to the used 120, render will give a seemingly nice image. But if you then look in Render /view/blur=none, you still see the locations of the spot on a squared (pixel??) grid. To me it seems the size of blurring increases with EM gain, but the locations found are still the same. I also changed pixel size and background, but still the same. 
2. played around with the box size: 7,11&13 all give the same pixelation (only visible with blur=none in render). 

So despite I said I found it working before, it seems like still the locations are pixelated (as with your average image), but for higher EMgain the blur is so large the pixels are blended into each other.

mwdocter, 2019-03-06:
to illustrate, the following pictures are made from the same raw file, parameters localize: box 11 pixels, EMgain 120, backgroud 400, pixel 107 nm. The only difference is the bluring in render/view, which is either global localization or none blur. 

[image]
[image]

straussmaximilian, 2019-03-06:
Hi, the camera settings will affect the blurring as this changes the calculated photon values and hence the localization precision. Can you try to see if it works with LQ? I have a hunch that maybe the images have low SNR which causes problems with the MLE. If possible, you could also share the raw file so I can see if I can reproduce the error.

mwdocter, 2019-03-06:
the raw file, uint8, 100 frames, 101x101 pixels, is in the following we transfer link https://we.tl/t-sTdPUHXCTQ
low SNR might indeed be the cause; in localise view/contrast I get values 5-39 for this uint8 data set. 
I just realized that this data set had already background subtraction before saving, so I changed the background in Localize to 0 (min net gradient 200), and finally am getting the subpixel positions. (not pixelized anymore)

Is there more information for how to set these values, other than given in the localize.rst?
The minimum net gradient is confusing, how can it find spots if the minimum net gradient of 200 is way above my maximum intensity?

joschnitzbauer, 2019-03-06:
Just chiming in here:

- I’d recommend collecting and processing movies with uint16. With uint8,
you might be losing relevant resolution on the intensity scale

- Localize should analyze data including background. Background subtraction
will seriously mess with all of the localization methods, because you
artificially cut the lower end of the intensity distribution. This is a
problem for all methods, but in particular for MLE where a certain
background distribution is assumed for the statistics.

- I suspect that the background subtraction and/or using uint8 causes the
pixelation artifacts in your data.

- While net gradient is roughly correlated with peak intensity, it is not
necessarily on the same scale. It’s the sum of intensity flowing to the
brightest point from all directions. So it can definitely be higher than
the peak intensity.

- For finding a good minimum net gradient cutoff, typically people check by
eye that most spots are detected while minimizing false detections. This is
obviously objective, but usually works well in some range. One could think
of more sophisticated approaches to determine cutoff automatically, but
most applications don’t need this.

Hope this helps!

mwdocter, 2019-03-06:
I so much agree with the uint16 instead of uint8, and still try to convince my colleagues who acquired the data. The uint8 conversion (data-bg/scalar) comes from scaling the data to be displayed on screen (which back then was limited to 8 bits). I'll pass on your arguements.

I'll make sure to double check baseline from now on, and try to find a good minimum net gradient cutoff.

Thank you again for your help

closed 2019-03-13 by straussmaximilian