jungmannlab/picasso #42: documentation FRET in Render

mwdocter, 2019-03-06:
In the Tools menu from Render, there are more items than the zoom/pick/measure documented in render.rst. The same for the postprocessing. 

I would like to do FRET on a data set with two channels displayed next to each other. 
I was hoping to be able to 
1. split the image into 2 (channels) and align them
2. automatically select spots (I guess with pick & pick similar)
3. extract fret traces from every spot (in 2 channels) (I guess with show fret traces, can I export the traces afterwards?)

Can you please give more documentation/ help me into the right direction on how to get FRET traces from already localized data?

straussmaximilian, 2019-03-07:
Hi,
this is correct - there are still some undocumented features.
The FRET implementation was used for this study: https://pubs.acs.org/doi/10.1021/acs.nanolett.8b02185 Here, the FRET efficiency is calculated based on single-molecule localizations. 

The idea is that you already have two separate movie files that are localized (if you have them next to each other I would probably use ImageJ to split them). The rest is pretty similar as you expected:

1. Align via Align channels
2. Select spots via pick and pick similar
3. Examine traces via show fret traces, use Calculate FRET In picks to calculate FRET values for all picks

Each localization will get a FRET value so that you can reload them and color code them via render properties.

mwdocter, 2019-03-07:
After opening two (donor& acceptor) channels, I first aligned the channels.

Then I started undrift by RCC. I have to pick a channel. Is it true that the drift is calculated from the chosen channel, and the undrift is applied to both channels?
Or is the drift only applied to the chosen channel (which seems illogical to me)?

mwdocter, 2019-03-07:
as follow up, it actually seems like undrift by RCC works on only one channel. In my case I had the two channels on the same camera, so the drift is the same. Is there a way to export and apply the drift found from the donor channel onto my acceptor channel?

mwdocter, 2019-03-07:
after drift correction and align channel, I picked spots and picked similar spots (in total 1012). Then I try to calculate FRET form picks, which results in an error:
[image]

Is there a limit to the number of picked regions?

straussmaximilian, 2019-03-08:
Hi,
the drift is calculated for each channel separately and only applied to the selected channel. This is consistent with the idea of recording consecutive datasets as in Exchange-PAINT. 
The calculated drift is automatically saved to files ending with "_drift.txt". It is possible to apply drift to files via txt files using the "picasso undrift -f _drift.txt dataset.hdf5" command. However, this command was written for another drift file definition so you can not use it directly but have to switch axis / +- first. I'll add this to the Wishlist list as I think this would be a useful feature.

There is no limit on the number of picked regions. I believe that this error is caused by selecting picked regions that do not have localizations in the acceptor channel. To avoid this, you can use the "Filter picks by locs" feature and filer picks so that they have at least one localization.

mwdocter, 2019-03-11:
I picked similar regions based on the acceptor channel. Then I filtered with "filter picks by locs" on both the acceptor channel with minimally 1 localization (also tried for 20). So for each point  I supposingly should have at least an acceptor signal. I am getting the same error still (list index out of range) when calculating FRET from picks.

Just to check which channel is the cause, I ran Calculate FRET from picks on only the acceptor channel (filled in acceptor twice), and then it runs without problems for minimal 20 localizations. For minimal 1 localization the error occurs. Same thing for twice donor channel with filter on 1 or 20 localizations. 

My best guess is that for certain spots I have an acceptor signal at a different time from the donor or vice versa. FRET should be IA/(IA+ID), so naively thinking, as long as I have either signal it should not be a problem. Is this true, or is it coded differently? 
And if so, why am I getting errors if calculate FRET for acceptor+acceptor channel at minimal localization 1? 

Side question: It would be nice to filter on both acceptor&donor channel minimal 1 localization, does that happen if I run the filters on both channels sequentially, or does the second filter overwrite the first filter?

straussmaximilian, 2019-03-11:
Hi,
I assumed that you used pick similar on the donor channel and hence need to make sure the regions also contain localizations in the acceptor channel. If you use pick similar on the acceptor channel, it is the other way around. 

Codewise, the FRET calculation is:
`fret_trace = acc_trace / (acc_trace + don_trace)`
If there is no acceptor signal for a given timepoint its zero, no donor its one. 

Having picks with only one localizations is an indication that the pick similar parameters are not chosen correctly and that you are picking up unspecific signal. 

Workflow-wise, proper settings for pick similar should ensure enough sampling for the first channel and the filer enough sampling for the second channel (be that acceptor or donor).

Each filter removes the picks that do not meet the criteria, so if you run them sequentially, the remainder of picks fulfills both criteria.

Hope that helps.

mwdocter, 2019-03-11:
For looking at FRET, I prefer pick similar on the acceptor channel. (and if possible, in sequence on the donor channel as well). I redid it now with minimally 100 picks on both acceptor and donor channel, and the third trace is giving me the list error. 
For tools settings I'm using diameter 3, pick similar +-2. 

Do you have more ideas I can try?

straussmaximilian, 2019-03-11:
Hi, it looks like the pick similar range is the default one and not adjusted. From the documentation:

Automated region of interest selection. Select ``Tools > Pick similar`` to automatically detect and pick structures that have similar numbers of localizations and RMS deviation (RMSD) from their center of mass than already-picked structures. The upper and lower thresholds for these similarity measures are the respective standard deviations of already-picked regions, scaled by a tunable factor. This factor can be adjusted using the field ``Tools > Tools Settings > Pick similar ± range``. To display the mean and standard deviation of localization number and RMSD for currently picked regions, select ``View > Show info`` and click ``Calculate info below``.

I assume that your picks have a rather high variance and little localizations, therefore a standard deviation of 2 could lead to 0 localizations as minimum values. If this should not work you could share the dataset with me and I could investigate.

mwdocter, 2019-03-12:
the data is on wetransfer: https://we.tl/t-cAQ7IRUfF5 

from my picked spots (I'm looking for the ones with dobth donor & acceptor signal) I calculated the info. 
For donor: 240 +- 278 localizations and RMSD to COM 0.53+-0.19
For acceptor: 550 +- 900 localizations and RMSD to COM 0.85+-0.2

In tools setting I can vary the diameter (cam pixel) and the pick similar range. With the found standard deviation, what would be a reasonable factor? It is a factor, and not the absolute standard deviation right? 
I tried 5 and, after filtering number of locatizations >100, I stilI get an error when calculating FRET values. I appreaciate it if you could have a look at the data (as outputted by localize, so still needs to be undrifted and aligned).

straussmaximilian, 2019-03-12:
For these values, I would set a standard deviation so small that the lower bound is at least above zero, i.e., 0.5 std -> For the acceptor the localization search range is then from 101 to 379.

I troubleshot your dataset and was able to reproduce the error. It seems that although there are a localizations in both channels, there is only little correlation in time. Attached is a sample trace;

[image]
 
I estimate that < 1% of the localizations for these datasets have both a signal in the acceptor and the donor channel at the same time. This leads to the error that a pick can have no FRET-events at all and hence runs into problems in the subsequent analysis. This, of course, is hard to troubleshoot if the FRET traces cannot be displayed in the first place and is something I had not encountered before. I now made the code a bit more robust so that the FRET functions can be called even though there are no FRET events present - show FRET traces will simply show no events in the FRET trace and calculate FRET in picks will raise an error message. 

Feel free to check out the version here and let me know if this works.
https://github.com/jungmannlab/picasso/releases/tag/v0.2.8

On another note, I noticed that your camera settings seem to be incorrect- I found a mean of 2.4 photons for the donor dataset and 1.8 photons for the acceptor dataset, which seems very low. The correct photon values are not only integral for the proper display of the localizations but could also lead to problems in various post-processing steps. 

Hope that helps.

mwdocter, 2019-03-13:
Indeed I did not pay much attention to the photon level, normally we are moslty interested in seeing two (relative) FRET levels, and not in the absolute FRET values. But if there is thresholding or other postprocessing steps which work with absolute values, then I will (have to) pay more attention to the correct photon levels.

The new update works beautifully, the program does not give anymore errors on my (admitted) not-the-best data set. The issue on getting better outcomes relies only on acquiring better data now.

straussmaximilian, 2019-03-13:
Great! I'll close this issue then.

closed 2019-03-13 by straussmaximilian

mwdocter, 2019-03-13:
#41 can be closed as well

Closed #42<https://github.com/jungmannlab/picasso/issues/42>.