jungmannlab/picasso #43: extracting individual traces from averaged traces

mwdocter, 2019-05-29:
Similar to the Nanoletters paper (2017, 17, 6428-6434) figure 2, I want to analyse our donor only origami construct in Picasso. After localise and render, an averaged image was retrieved.
From this averaged image I would like to extract time traces, either a summed intensity over the picked spot in time, or multiple time traces: one per aligned spot. 

Currently, the time trace shows the x-y position and ON/OFF over time, and does not take into account if at one time point multiple events occur, distributed over multiple spots. 

Is there a way to extract intensity time traces of picked regions in already aligned, but not yet averaged, spots?

straussmaximilian, 2019-05-29:
Hi,

> Similar to the Nanoletters paper (2017, 17, 6428-6434) figure 2, I want to analyse our donor only origami construct in Picasso. 

The analysis of Figure 2 was done using ImageJ on the raw data.

> After localise and render, an averaged image was retrieved.
From this averaged image I would like to extract time traces, either a summed intensity over the picked spot in time, or multiple time traces: one per aligned spot.

Currently, there is no option to sum and export the time trace of a picked spot. If the intent is to calculate an average FRET intensity, I would advise to analyze the individual FRET-events as it is done with the function Calculate FRET in picks. The summing of whole traces requires much more sampling due to their sparsity.

You can pick the region of interest and select `Save picked localizations`  to export the localizations within that pick. Each pick will be identifiable by the group information, so you could use that table to calculate the summed intensity or extract the time trace pick by pick.

> Currently, the time trace shows the x-y position and ON/OFF over time, and does not take into account if at one time point multiple events occur, distributed over multiple spots.

This is correct - `Show trace` shows each localization independently event and does not superpose events.

>  Is there a way to extract intensity time traces of picked regions in already aligned, but not yet averaged, spots?

If by alignment you mean the cross-channel alignment of donor-acceptor channel you can use the `Save picked localizations` as described above. 

Hope that helps.

mwdocter, 2019-05-29:
As suggested, I will try the save picked localization to get summed time traces pick by pick. 

I still think it might be usefull to extract traces from the same sub region of aligned picks. But for now, I will stick to comparing time traces of the full picked region.

closed 2019-05-29 by straussmaximilian