jungmannlab/picasso #44: Bug in Saving Picked Regions to YAML file

chaosunlucas, 2019-08-21:
Hi, I recently ran into this problem that was not there before: I can successfully save picked regions in Picasso:Render to a YAML file and use it in Picasso. But when I open it as a text format, it fails to show the pixel coordinates of picks. Instead, it shows below. Is there any fast solution to this? I need these coordinates for some downstream analyses in MatLab and now I cannot import them. 

**Centers:
- - !!python/object/apply:numpy.core.multiarray.scalar
    - &id001 !!python/object/apply:numpy.dtype
      args: [f8, 0, 1]
      state: !!python/tuple [3, <, null, null, null, -1, -1, 0]
    - !!binary |
      zczMzMxsQEA=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      PQrXo3CNYEA=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      16NwPQo3QkA=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      j8L1KFyHd0A=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      ZmZmZmaGQ0A=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      pHA9Ctcjd0A=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      XI/C9Si8Q0A=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      7FG4HoVXd0A=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      cT0K16MwRUA=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      rkfhehQ2d0A=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      rkfhehTORkA=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      4XoUrkfVdkA=
- - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      CtejcD2qS0A=
  - !!python/object/apply:numpy.core.multiarray.scalar
    - *id001
    - !!binary |
      MzMzMzMzd0A=**

straussmaximilian, 2019-08-22:
Hi,
it seems that there is a difference in the way a float and a float64 is represented. I found this:
https://stackoverflow.com/questions/40691311/save-results-in-yaml-file-with-python

For a quick solution, you could potentially re-read the file in python and save it again. I could potentially provide you a little script for that - let me know if you need assistance.

Other than that: Could you tell me the steps that lead to this error? Then I can troubleshoot it in the code directly.

chaosunlucas, 2019-08-22:
Hi, thanks for the reply.
I loaded a YAML file generated by MatLab to Picasso:Render to identify specific regions of interest. The picked regions were adjusted in Picasso:Render before I saved the edited picked regions again as a YAML file. This new YAML file still works for Picasso but does not open correctly as text.file anymore.

However, this problem seems to be gone when I redid the process so I am not entirely sure which steps were special last time. The picked region coordinates usually have two digits, but, often, at the end of the list the digits can go up to ~10 digits. I wonder if this could be the problem for this error: that somehow the number of digits got too high to be opened correctly as text.file

straussmaximilian, 2019-08-27:
Great that it worked. I changed it now so that the values should be cast as non-numpy float, see commit #7813b75. Let me know if the bug should pop up again.

closed 2019-08-27 by straussmaximilian