jungmannlab/picasso #47: Errors in Undrift by RCC

chaosunlucas, 2019-09-20:
I recently encountered a couple 'novel error's when doing RCC drift correction. It would be very helpful to know what cause these errors and whether there are solutions to it. Thanks!

1. operands could not be broadcast together with shapes (25,)(15,)
2. zero-size array to reduction operation maximum which has no identity

straussmaximilian, 2019-09-20:
Hi,
recently no changes in the RCC were made. Typically a failure in the RCC is dataset related. However, this should show an error message.
Which version of Picasso are you using?
Also, if you can share the dataset and upload it I could investigate.

chaosunlucas, 2019-09-24:
Below is the dataset. Thanks for investigating.

https://drive.google.com/file/d/1cuL5W9CRrp0qo5Nw_-ecz7pLgPrgr4z9/view?usp=sharing

straussmaximilian, 2019-09-24:
Hi,
it seemed like that the sampling was not sufficient for the drift correction algorithm. See a sample trace below. I now changed the algorithm to always run through. The default drift for a segment is now 0,0. 
However, it is to note that for sparse sampling the drift correction with RCC is very likely to not produce the desired results. This can be identified by a high variance in the drift plot.

[image]

closed 2019-09-24 by straussmaximilian