jungmannlab/picasso #51: Cannot make v0.2.8 work on W7

PierreMangeol, 2019-10-22:
* Picasso version: v0.2.8 (from Windows installer)
* Python version: 2.X and 3.7 installed (I guess this doesn't matter)
* Operating System: Windows7

### Description

Hi,
I wanted to try your software. I used Installer 0.2.8 - Windows 64-bit (I guess this would lead to stand-alone application?) to have it on my computer.
Whatever subprogram I try to launch (Design, Simulate...) I get the following message "Failed to execute script picasso-script".

Any idea what can cause this and how to solve it?

Thank you for your help!
Kind regards,
Pierre

straussmaximilian, 2019-10-22:
Hi Pierre,
Sorry to hear that the installation is not working.
I just tested the installer on a Windows 10 machine and everything was working as intended. I would suspect that this is related to Windows 7.

Unfortunately, I don't have a Windows 7 - Machine around that I could use to test and troubleshoot. I think it also wouldn't make too much sense in investing a lot of time getting this to run - Windows 7 is now 10 years old and Microsoft will drop the support in early 2020.

So my advice would be to try one of the older releases (I noticed that some of the links to the installers were not working - now they should), or if you have Python installed anyhow use the Python version.

I hope that helps. 
Best,
Max

PierreMangeol, 2019-10-22:
Hi Max,
Thank you for replying fast!
I found a computer with Windows 10 and it works well. Thanks for making the older versions available (indeed links were broken earlier today), this is definitely useful.
I fully understand that maintaining Windows 7 versions does not make a lot of sense.

Thanks for your help.

Kind regards,
Pierre

closed 2019-10-22 by PierreMangeol