jungmannlab/picasso #86: Code adaptations to avoid error messages from pycharm

hartlbirgit, 2020-03-09:
* Picasso version: Picasso-Windows-64bit_0_2_8
* Python version: 3.7, Anaconda distribution
* Operating System: Windows 10 64bit

### Description

I created a project in PyCharm and imported simulate, localize and render
PyCharm reported missing imports but I could run the three programs, but Python crashed within the debugger when calling render.
Here is my calling code (I omitted sys.exit() at the end of each module, to be able to call them consecutevely)

from picasso.gui import simulate
from picasso.gui import localize
from picasso.gui import render
simulate.main()
localize.main()
render.main()

### What I Did

Since PyCharm reported a missing imports and erros, I imported in "picasso/gui/simulate":

import traceback
from PyQt5.QtWidgets import QDialogButtonBox, QDialog
from PyQt5 import Qt

and in "picasso/simulate"
I casted in function "generatePositions(...)" variable "spacing" to int to avoid an exception

and in "picasso/gui/render" I imported
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
(instead of
from matplotlib.backends.backend_qt5agg import FigureCanvas as FigureCanvas)

```
Thus I could avoid the crash of Python within debugger when calling render

I do not know whether I have done eveything correctly, since I am very new to Python, but I hope this contribution is helpful
```

straussmaximilian, 2020-03-09:
Hi,
thank you very much for reporting this issue and providing your solution!

To easily integrate your code changes, GitHub allows making Pull Requests. This way It is a bit easier to see where the changes were made in the code. You can find an introduction to how to do this [here:](https://github.com/jungmannlab/picasso/blob/master/CONTRIBUTING.rst)
Note that the pycodestyle and the tests are currently not up to date, so I wouldn't worry about this for now.

If you want to feel free to give this a try - this way you would also be automatically listed as a code contributor in GitHub. If not that is also fine, let me know and I can simply copy and paste the code you mentioned above.

Thanks again!

hartlbirgit, 2020-03-10:
Hallo,
thank you for your proposal how to proceed. But in this case I would prefer that you enter the code changes. Strangely enough "render" seems to run stable from an anaconda prompt, without the import change, but simulate definetively lacks the cast to "int" within generatePositions(...), what is reported within anaconda prompt by "traceback" that is reported by PyCharm as missing import.
This leaves me somewhat puzzled though...
Best

straussmaximilian, 2020-03-10:
Hi,
yesterday I tested all the modules which I could run from the terminal w/o the import changes.
As for simulate I also did got the error. I guess this was caused by a package update.  Unfortunately, there is no full coverage of tests at this moment so that those kinds of errors can occur when trying to keep the packages up-to-date w/o being noticed.  I fixed this with commit #7cc2dd9.

closed 2020-03-23 by straussmaximilian