jungmannlab/picasso #91: Localize freezes when using regions from picks

GolgiWhillikers, 2020-03-27:
* Picasso version: 0.3.0
* Python version: 3.7
* Operating System: Windows 10

### Description

Localize freezes with a "Cannot set parent, new parent is in a different thread" error (see image) in one specific instance: I am trying to do a 3d calibration using picked regions as described elsewhere in the documentation. I localize an astigmatic bead stack, pick the beads in Render, then use the identified picks again in Localize to do the 3d calibration. I am using LQ+GPU fit. When I then click on "calibrate 3d" I get this error. If I do exactly the same thing but do not try to load the picks, I do not have an error and the calibration proceeds normally. The same error occurs with MLE instead of LQ. I just did a fresh install of Picasso (this is on a new computer) this morning, so this should be all the latest Anaconda/Picasso/Python package versions on a Windows 10 machine. Let me know if there's other information I can provide. Thanks

[image]

### What I Did

```
See the attached image for the error and traceback in anaconda prompt.
```

straussmaximilian, 2020-04-10:
Hi,
The bug was related to having picks that would be out of bounds with respect to the box size. Everything OOB will now be removed. If possible, could you check if the error persists?

GolgiWhillikers, 2020-05-14:
Hi, sorry that I am just getting around to checking this. I am still having issues, but they are new and perhaps related to updated files.

I updated Picasso with `git pull https://github.com/jungmannlab/picasso.git`, which went seemingly fine. However, at first I could not launch anything because hdbscan was missing. `pip install hdbscan` was giving me an error for not being able to build the wheel file due to a PEP517 issue...though after a little googling I was able to add it with `conda install -c conda-forge hdbscan`. That did allow things to open. Not sure if this is an issue for other users.

In terms of the specific bug, I am now running into a new one. When I load the localized bead stack into render, I can make picks fine, but then when I try to save them with File>Save Pick Regions, Render freezes. It is not throwing an error. Actually, I'm now noticing that all the File menu options are freezing. Other menus like Postprocess seem to work OK (Undrift by RCC worked, and making pics worked), but I cannot save anything. And actually, this appears to be some sort of universal file menu thing, because Filter is also hanging when I try to save, as is Localize if I choose anything from the File menu, including the aformentioned "Load Picks as Identifications".

I tried running `python setup.py install` in case the new version of Picasso had bits that git pull didn't install (I'm a novice with python and git), and this did not solve the problem. Otherwise, the only changes I made in this Anaconda environment from the previous version were the git pull and installing hdbscan.

straussmaximilian, 2020-05-24:
Hi, thanks for reporting the issue.
In order to also upgrade your environment after pulling the latest commit, you can use the following command:
`pip install --upgrade -r requirements.txt`
I tried to reproduce the Save Pick Regions error and also experienced a freeze bug, which should now be fixed with 2ba3bc7
 
I also tried saving some localization files in filter after filtering, and this worked fine. 
Could you maybe pull the latest version, try the pip command and see if this error persists?

GolgiWhillikers, 2020-05-27:
OK so some success, some further problems:

1. Ran the pip upgrade command as you suggested. This seemed to work fine, but then when I tried to open Localize, I got an error that Numba was unable to be loaded (see attached image). On some googling, it seemed like this was an issue with Numba/Python version mismatch. However, I was worried that my picasso env had gotten messed up, so I deleted and did a clean install of Picasso and associated packages with Python 3.7.

2. On clean install, `pip install -r requirements.txt` did not work as hdbscan did not install correctly by PIP (as above). However, if I installed hdbscan ahead of time as I mentioned above, the requirements.txt worked fine. Either way, this did not fix the numba import issue. I tried setting up a second environment with Python 3.8, but this did not fix either the hdbscan install behavior or the numba import issue. Finally, I downgraded numba to 0.48.0 in the python 3.7 env, and this allowed Picasso modules to launch as expected.

3. After setting up the env, I tested the workflow. Loading the RAW image into Localize and localizing spots worked fine. I then loaded the resulting hdf5 into Render, picked regions, and saved them, which now worked did not cause a freeze. Then, I re-opened localize, loaded the original RAW file, and chose "Load Picks as IDs". Picks did load (no freeze), but, only a fraction of the picks actually loaded - ie, I picked 6-7 regions, and 2-3 loaded. This is new behavior.

Numba error
[image]

Render picks
[image]

Localize fails to load all picked regions
[image]

straussmaximilian, 2020-06-12:
Hi, I did some updates to the codebase:

- Updated installation instructions on the readme, rely only on requirements.txt
- Removed the hdbscan requirement as it caused problems on Windows machines, this is now optional.
- Updated to Python 3.8 and the latest packages.

I created a new environment based on the new instructions, tested to localize files, load them in render, export the picks, and load them in localize again, which worked.
Could you try to reproduce the error?

GolgiWhillikers, 2020-08-04:
Hi-
Sorry I am just getting around to this with pandemic science time.

-Thanks for the updated install/update instructions, that makes it much easier.
-I pulled the latest update, then created a new environment with Py3.8 and installed using the instruction the readme and requirements.txt only. This worked fine. hdbscan module failed to install as before.

However, the behavior persisted for me. Drag RAW file in, localize (had some issues with GPUfit but that's maybe because I'm remoting in; it worked fine without GPU), load hdf5 into render, pick 10 regions, save regions, drag RAW back into localize, load picked regions...and only a few of those 10 regions load.

So I'm not sure what to try next. Delete everything and try to fully install from scratch? Maybe I have some weird package installations? This is all on Windows 10 in conda 4.8.3. I'm on the most up to date version of picasso as far as I can tell (git pull says already up to date; I'm not sure how to tell what specific version has been pulled)

I've got a separate issue with render that I'll post a separate issue about in a minute.