jungmannlab/picasso #124: Problem with Render module render scale

GolgiWhillikers, 2020-08-04:
* Picasso version: most recent pull, looks like v 0.3.1
* Python version: 3.8, and also 3.7, it's an issue in both
* Operating System: Windows 10

In the Render module, when I chose to render by some parameter, say, Photons, the actual rendering will work fine, but then if I try to open the scale bar for the color code, Render will freeze and crash.

1) Open Render
2) Drag in hdf5 file
3) Zoom to some region
4) Choose view>display settings, then in render properties, choose photons, click render. Rendering proceeds correctly.
5) Click show legend
6) The legend window pops up, but is empty. I can still interact with the render module, but if I try to interact with that window, Render hangs and crashes as below, with the messages in the picture in the anaconda prompt

This function has worked in the past for me (maybe a year ago) but I think it cropped up around the same time there were issues with the figures popping up for drift correction et al as well with the PyQt5 switch.

[image]
[image]

Thanks

straussmaximilian, 2020-10-20:
Hi,
I just updated all packages and tried to reproduce the error (Python 3.7.8), which I couldn't.
Can you try again?

[image]

This was installed in the env:

asteval==0.9.19
certifi==2020.6.20
colorama==0.4.4
cycler==0.10.0
future==0.18.2
h5py==2.10.0
joblib==0.17.0
kiwisolver==1.2.0
llvmlite==0.34.0
lmfit==1.0.1
matplotlib==3.3.2
mkl-service==2.3.0
numba==0.51.2
numpy @ file:///D:/bld/numpy_1603047752645/work
perfplot==0.6.2
Pillow==8.0.0
pyparsing==2.4.7
PyQt5==5.15.1
PyQt5-sip==12.8.1
python-dateutil==2.8.1
PyYAML==5.3.1
scikit-learn==0.23.2
scipy==1.5.3
six @ file:///home/conda/feedstock_root/build_artifacts/six_1590081179328/work
termtables==0.1.0
threadpoolctl==2.1.0
tqdm==4.50.2
uncertainties==3.1.4
wincertstore==0.2

GolgiWhillikers, 2020-11-18:
Apologies for the slow reply. I can confirm that this fixed the problem.

-Updated Picasso by the main page directions in a python 3.8 env (picasso version 0.3.1, I guess it's dde20a9?)
-Followed my original post to reproduce the error
-Legend window appeared as expected (though, it'd be nice if it would scale the text, but hey, at least it's there!).

Thanks

[image]

closed 2021-07-31 by straussmaximilian