jungmannlab/picasso #140: Automating "Undrift from pick" procedure for headless systems

seb45tian, 2020-10-16:
* Picasso version: 0.3.1
* Python version: 3.7.2
* Operating System: CentOS 7.8

### Description

Is there a way to use the **Undrift from pick** procedure without the need to interact with the GUI, e.g. as a Python library?

### Background

I'm in HPC support and have a user trying to **undrift** a high-res image (~2.5G) from 10k up to 50k picks. This causes the need for large amounts of Memory (e.g. with 20k picks, around 120G are needed). We provide nodes with sufficient amounts of memory but no option for GUI interactions (i.e.  a headless system). Therefore a scripted, CL driven solution would be beneficial. (Maybe even for other processing features?)

### Tested (non-working) solution

My idea was to use the functions responsible for the undrift procedure directly from the gui/render.py library, i.e.

  - File:Open -> Window Class -> open_file_dialog()
  - File:Load pick region -> View Class -> load_picks()
  - Postprocess:Undrift from pick -> View Class -> undrift_from_picked()

However trying this in a python script, e.g. like this

```python
from picasso.gui.render import View

view = View()
```
of course crashes, as all the Qt Window backend stuff is missing. Is there anything usable for this case, or would one have to re-implement this functionality from scratch?

straussmaximilian, 2020-10-20:
Hi,
Thanks for your suggestion.
A good starting point would be `__main__.py`, which has the `localize` option to batch process files via command-line. It already has the undrift via the rcc option.

Implementing something like undrift from picked would require some tinkering but should be feasible.

Essentially you would need to call the `_undrift_from_picked_coordinate` from `gui\render` function. This requires to have `picked_locs` from the `picked_locs`-function. This should not require Qt functions.

If you want to automate the picking process, you would need to estimate some starting conditions (what else is done via the initial picking); else this should also be able w/o Qt (see `pick_similar`)

This might also be useful for the auto-picking:
https://github.com/schwille-paint/picasso_addon

I hope that helps!

rafalkowalewski1, 2024-07-05:
In Picasso release 0.7.0 AIM (https://www.science.org/doi/10.1126/sciadv.adm7765) will be implemented. Undrifting then can be run without picking and with good temporal resolution. I think this should solve the problem with the manual input for undrifting.

Best wishes,
Rafal

closed 2024-07-05 by rafalkowalewski1