jungmannlab/picasso #150: Picasso not starting on Windows

timojuez, 2020-11-12:
* Picasso version: 0.2.8
* Python version: ?
* Operating System: MS Windows 10

### Description

Open Picasso

### What I Did
I installed Picasso from the standalone installer.
picasso.exe shows a help message and picassow.exe exists immediately without any message. How can I start it?

ajasja, 2020-11-18:
I don't think there is enough info to help debug. At least an error message would be needed. 
If you run picasso.exe from the console, what do you get?

You could try installing using conda (works for me on windows).

timojuez, 2020-11-19:
Directly after install, Picasso.exe printed an argument help and picassow
quit without any message. Now (maybe because of reboot) it is this:
"C:\Program Files\Picasso\picasso.exe"
Traceback (most recent call last):
  File
"site-packages\pyinstaller-3.3.dev0+gb78bfe5-py3.5.egg\PyInstaller\loader\pyiboot01_bootstrap.py",
line 18, in <module>
  File
"C:\Miniconda3\envs\picasso\lib\site-packages\pyinstaller-3.3.dev0+gb78bfe5-py3.5.egg\PyInstaller\loader\pyimod03_importers.py",
line 664, in install
  File
"C:\Miniconda3\envs\picasso\lib\site-packages\pyinstaller-3.3.dev0+gb78bfe5-py3.5.egg\PyInstaller\loader\pyimod03_importers.py",
line 224, in __init__
ImportError: Can't load frozen modules.
Failed to execute script pyiboot01_bootstrap

ajasja, 2020-11-20:
Strange. I would try to install somewhere that is not in the program files, but in a folder that you have write access to (like C:\tmp).

How familiar are you with conda?
If you install minicinda (https://docs.conda.io/projects/conda/en/latest/user-guide/install/) and then install picasso into the conda environment as described on the main page then it should definitely work.

straussmaximilian, 2020-11-21:
Unfortunately, at this point, the one-click installer is broken and not up-to-date; you could try installing an older version or using the python condo installation as @ajasja suggested.

closed 2021-04-15 by straussmaximilian