jungmannlab/picasso #153: Whish list: zoom in and out with the mouse wheel.

ajasja, 2020-11-17:
* Picasso version: dde20a9ff
* Python version: 3.8
* Operating System: Windows 10 (20H2)

### Description

It would be nice if Ctrl + mouse wheel would zoom (like in Fiji). The mouse position could control the zoom center. 

PS: Picasso is great.

straussmaximilian, 2020-11-21:
Hi,
I just pushed a commit 270f7be that should contain this functionality.
When now pressing CTRL and using the mouse wheel, the zoom function will be activated. The zoom center is taken from the mouse position. However, this will mean that one needs to adjust the mouse position when zooming further, e.g., the object of interest is in the upper right corner, use the mouse wheel, zoom is applied, the object is now in the center position, to zoom further mouse needs to be moved to the center position.

Let me know if this is the intended usage or if you have additional suggestions.

ajasja, 2020-11-24:
Thank you, this is great! This is exactly the suggested usage.

closed 2020-11-24 by ajasja