jungmannlab/picasso #154: Opening raw file in the localize module.

ajasja, 2020-11-18:
* Picasso version: dde20a9
* Python version: 3.8
* Operating System: Linux 64 bit

### Description

I'm trying to open a raw file in the localize module, but raw files are not shown in the open dialog.
[image]

### What I Did

```
python -m picasso localize
File->Open Movie
```

ajasja, 2020-11-18:
Fix in https://github.com/jungmannlab/picasso/pull/155

Also workaround is to drag & drop files.

straussmaximilian, 2020-11-21:
Thanks for reporting the issue and also coming up with a fix. This is very much appreciated!

closed 2020-11-21 by straussmaximilian

ajasja, 2020-11-24:
You are very welcome! It was luckily an easy fix:)