jungmannlab/picasso #192: crash when opening picked localizations in Render

pnickels, 2021-04-07:
* Picasso version: 0.3.1
* Python version: 3.8
* Operating System: Win10, MacOS 11.2.3, Ubuntu 18.04 LTS

### Description
When I try to open picked localizations in the Render module, I consistently get the attached "IndexError: list index out of range" message. I initially thought it was a problem with the Picasso installation on our Linux server, but I get the same error on Windows and Mac systems.

[image]

straussmaximilian, 2021-04-15:
Should be fixed now. Looked like the bug only appeared when using the File menu and not the drag and drop.

closed 2021-04-15 by straussmaximilian

pnickels, 2021-04-16:
Thanks Maximilian!

commit 491ae289 (2021-04-15): FIX #192