jungmannlab/picasso #193: different fits of same data with and without GUI

SimoneEzendam, 2021-04-23:
* Picasso version: 0.3.0 (and/or 0.2.8)
* Python version: python-3.7.4
* Operating System: Win10

### Description

I am trying to make my life easier by integrating the localization into the rest of my analysis code (omitting the GUI, as far as I can see the newest version there is 0.2.8). 

### What is wrong
I get the localization to run, but am unable to reproduce the results I get from the GUI. There is usually some localizations that are identical in both, but most are different. I will attach an example below.  

### What I Did

 I adjusted the run function from the FitWorker class to work without the GUI (and only for MLE with "sigmaxy"), but don't see where the different results can arise.

##### code
```
import time
import pandas as pd
import h5py
from picasso import gaussmle
from picasso import localize
from picasso import io

def run(movie, identifications, box, camera_info): #minimized version of the run function inside the FitWorker class from picasso.gui.localize
    N = len(identifications)
    t0 = time.time()
    spots = localize.get_spots(movie, identifications, box, camera_info)

    curr, thetas, CRLBs, llhoods, iterations = gaussmle.gaussmle_async(
        spots, eps=0.001, max_it=1000, method="sigmaxy"
    )
    while curr[0] < N:
        time.sleep(0.2)
    locs = gaussmle.locs_from_fits(
        identifications,
        thetas,
        CRLBs,
        llhoods,
        iterations,
        parameters["Box Size"],
    )
    #self.progressMade.emit(N + 1, N)
    dt = time.time() - t0
    #self.finished.emit(locs, dt, self.fit_z, self.calibrate_z)
    print('fitted in ' +str(dt) +'s.')
    print(curr, thetas, CRLBs, llhoods, iterations)
    return locs

def save_locs(path, df): #need to copy a yaml file manually for now
    locs = df.to_records(index=False)
    with h5py.File(path, "w") as locs_file:
        locs_file.create_dataset("locs", data=locs)

path = 'filepath.tif'
#camera info
camera_info = {}
camera_info["baseline"] = 0
camera_info["gain"] = 200
camera_info["sensitivity"] = 3
camera_info["qe"] = 0.97
camera_info["pixelsize"] = 163
#parameters
parameters = {}
parameters["Box Size"] = 7
parameters["Min. Net Gradient"] = 15000

movie, info = io.load_movie(path, prompt_info=None)
identifications = localize.identify(movie, minimum_ng=parameters["Min. Net Gradient"], box=parameters["Box Size"])
locs = run(movie, identifications, parameters["Box Size"], camera_info)
locs_df = pd.DataFrame.from_records(locs)
save_locs('loc_file.hdf5', locs_df)
```
#### same issue with just picasso.localize (no adapted run function)
yet different localizations (again, some are the same) when I use 
##### code
```def localize(movie, camera_info, parameters): 
    print("localizing")
    identifications = identify(movie, minimum_ng = parameters["Min. Net Gradient"], box = parameters["Box Size"]) # change from Simone, parameters)

    return fit(movie, camera_info, identifications, parameters["Box Size"])
```
### Any help/suggestions would be greatly appreciated :)

[image]

SimoneEzendam, 2021-04-23:
I just verified that the current version and version 0.2.8 give essentially the same results from the GUI
[image]

straussmaximilian, 2021-04-23:
Hi,
Thanks for reporting this.
What settings are you using in the GUI? The default min gradient is 5000 (vs. the 15000 in the script), so maybe this could cause differences, and I would double-check this. The same would be for the camera settings.
If they are identical, do you have a test file that you could upload so I could test this?

I hope this helps.
Best.

SimoneEzendam, 2021-04-23:
latest version from (right)GUI and batch processed (left) from the cmd with the same parameters. x & y are similar (they differ a bit), the photons (and I am assuming therefore the other parameters) change

[image]

SimoneEzendam, 2021-04-23:
thanks for getting back so soon! The camera settings & min net gradient are the same.. Can't upload it here, unfortunately, but here is a link: https://cloud.physik.lmu.de/index.php/s/craCfCaJ676crxQ

straussmaximilian, 2021-04-24:
Hi,
It looks like there were some issues in the way that the MLE fitting was called.
What you want to do is call `fit_async` and not `gaussmle_async`.
I hope that helps.

closed 2021-06-25 by straussmaximilian