jungmannlab/picasso #199: Processing file bigger than 4GB on Picasso Localize

gwisna, 2021-05-14:
* Picasso version: 0.2.7
* Python version: 3.8
* Operating System: Windows 10 enterprise

### Description

I need to process the localization for file bigger than 4GB, is there a way to process it with Picasso localize? I do not want to crop my movie that contains 90000 frames. 

I look forward for some solution that I can try.

```

straussmaximilian, 2021-06-25:
Hi,
in priniciple there should be no file size limit. Did you experience any error with your file?

scholer, 2021-08-08:
Is this a TIFF file or RAW file? The original TIFF format only supports files up to 4 GB. If you have a TIFF file larger than that, it is in the new "BigTiff" format which is not yet supported by the main Picasso (at least, it wasn't when I last checked 2 years ago). I have made a fork that implements support for BigTiff format and reads TIFF files larger than 4 GB without issues. I wanted to submit a PR into the main branch, but I just haven't had time to polish it yet. You can find here: https://github.com/scholer/picasso/tree/bigtiff

Alternatively, you can also convert your large TIFF file to .RAW format and load that with Picasso instead.

Antimonyleo, 2022-12-16:
Hi, is there any plan to implement the file reader to support "BigTiff" any time soon? I have been generated a lot of (huge) datasets and it would be a lifesaver if I don't have to convert the .tif file to .raw (which also eats up my storage space very fast as now I am doubling the size of each dataset). Thanks!

rafalkowalewski1, 2022-12-17:
> Hi, is there any plan to implement the file reader to support "BigTiff" any time soon? I have been generated a lot of (huge) datasets and it would be a lifesaver if I don't have to convert the .tif [...]

We are planning to include this file format too, however, this will come in further releases.
Bests,
Rafal

IrrelevantNameTom, 2023-04-27:
Hello,
Using micro-manager 1.4 to record data, the image is split into  4Gb files but Picasso Localize will still open all the frames. Maybe it is useful to you if you cannot use .raw conversion for any reason.
Best,
Tom

ajasja, 2024-05-29:
I would also appreciate bigTiff support. We have files about 4.4 GB in size and the error is: 
[image]

@IrrelevantNameTom Do you know what the naming of the files has to be, for picasso to load all the frames?

ajasja, 2024-05-29:
Ok, figured it out 
[image]