jungmannlab/picasso #203: Opening TIF files larger than 5GB in Picasso Localizer

FrejaPi, 2021-06-18:
* Picasso version: From april 15th
* Python version: 3.8.10
* Operating System: Windows 10

### Description

I am trying to open TIF files, but are getting the error seen below:

[image]

### Further comments
RAM is not the issue. 
Also, I am able to open files that are 4,5GB, acquired from the same computer, as the larger files had been acquired on.

I have tried both dragging and opening through the file tab, but it does not make any difference.

In advance, thank you for your help

straussmaximilian, 2021-06-25:
Hi,
this looks like a problem with the type of TIF file. 

Picasso is specifically tailored for ome.tif files that are created when using MicroManager. Generic TIFs could cause all sorts of problems.
The workaround for this is to use the exporter script for ImageJ (see https://github.com/jungmannlab/imagej-raw-yaml-export). This will convert the image to a raw file and a corresponding yaml file so that Picasso can read it.

Did you use a MicroManager TIF file?

FrejaPi, 2021-06-28:
Hi,

The files were not created using MicroManager, which seems to be the problem. However, I tried the plugin for ImageJ, and this works for my files.

Thank you for your help

Kind regards, Freja

straussmaximilian, 2021-06-28:
Awesome! Glad that it works.

closed 2021-06-28 by straussmaximilian