jungmannlab/picasso #222: GPUfit option not available using picassosr Python package

GolgiWhillikers, 2021-11-02:
* Picasso version: picassosr 0.3.7
* Python version: 3.8
* Operating System: Win 10

When I install picasso from github and launch programs from the picasso folder using `python -m picasso localize` for example, the "Use GPUfit" option is available and works. However when I install picasso as the picassosr python package with pip, the "Use GPUfit" option is greyed out and unavailable.

pygpufit is installed in the environment
[image]

but GPUfit is unavailable
[image]

I installed pygpufit in this environment the same way I've installed it in the past for github installations. Am I missing a step somewhere to make it available? 

Thanks

straussmaximilian, 2021-11-04:
Should be fixed now. Currently pushing the new version to [PyPi](https://github.com/jungmannlab/picasso/actions/runs/1421985303) - do you mind to test if this solves the problem?

GolgiWhillikers, 2021-11-04:
I ran `pip install picassosr --upgrade`, upgraded fine to 0.3.8. GPUfit box is no longer greyed out.

However I ran into an error when it tried to do the GPU fit on my test region. The first time, python just hung. The second time, I got the following error message:

[image]

And in the anaconda prompt:

[image]

GolgiWhillikers, 2021-11-04:
I'm actually getting the same error using python installed via github (ie launching with `python -m picasso localize`) and using an older version of picasso (0.3.1), on a totally different computer. I updated to 0.3.8 via git pull and saw the same behavior. In this case, I'm using pyGpufit 1.1.0 due to having an older GPU. Same problem with 0.3.8 installed in this way.

straussmaximilian, 2021-11-04:
Hi,
it looks like that `time.clock` is depreciated in [Python 3.8](https://stackoverflow.com/questions/58569361/attributeerror-module-time-has-no-attribute-clock-in-python-3-8). Do you mind trying downgrading your Python environment? As this is a pygpufit-issue I will create an issue there.

straussmaximilian, 2021-11-04:
Hi, so I just checked.
The current version in gpufit should not have this issue and was closed already some time [ago].(https://github.com/jungmannlab/picasso/issues/139)

Do you mind trying the latest gpufit version and install?

GolgiWhillikers, 2021-11-04:
Oh OK thanks - I uninstalled pygpufit 1.10 and installed pygpufit 1.2.0 in both my picassosr and git pulled environments (both picasso 0.3.8 I believe), and GPU fit worked correctly in each. Thanks.

straussmaximilian, 2021-11-04:
Awesome!

commit 977d21d5 (2021-11-04): Fix #222
closed 2021-11-04 by straussmaximilian