jungmannlab/picasso #227: Define mag factor and 3d calibration path with localize in CMD

GolgiWhillikers, 2022-01-03:
* Picasso version: picassosr 0.3.8
* Python version: 3.8
* Operating System: Win 10

Is there a way to programmatically include the magnification factor and 3d calibration yaml path when calling picasso localize from windows command line? My understanding is as of now, if a 3d localization method is called, the program will ask you to manually input the mag factor and 3dcalib.yaml path. I'd like to be able to provide these as function inputs so the user doesn't have to provide them and we can just loop through a bunch of folders. We are trying to batch process files where we need to split an image acquired with a DualView, calibrate each side separately, and then fit each side separately with the appropriate calib file, so it would be very useful to designate the file path and mag in the function input from CMD. It's certainly possible I'm missing an existing way to do this, but I don't see it as an input to localize in the CMD documentation. We could probably do folder wrangling to put the appropriate calib files in a folder with the appropriate half-TIFFs, but a direct input would be easier. Thanks

straussmaximilian, 2022-01-29:
Hi, i added this to 0.3.9, you can now add the path via -zc and the magnification factor via -mf. Let me know if this works!

GolgiWhillikers, 2022-02-03:
Awesome, I tested this on my build with pip install picassosr --upgrade and it works for me. Thank you.

closed 2022-11-20 by GolgiWhillikers