jungmannlab/picasso #229: Wishlist: Localize within ROI

GolgiWhillikers, 2022-01-26:
* Picasso version: picassosr 0.3.8
* Python version: 3.8
* Operating System: Win 10

I'd love to see functionality implemented to localize within an ROI in the Localize GUI (and/or a defined ROI coordinate in the cmd line input). It's particularly useful for images acquired on a multiview system, ie simultaneous red and far-red emission, where each side needs to be localized with different parameters. Often the background is different between the two channels necessitating different net gradients, or the 3d calibrations for the 2 sides are sufficiently different due to chromatic aberrations to need different calibrations. Currently, one needs to either localize the same field twice and crop after the fact with Filter or otherwise which can be time-intensive for large files, or one needs to pre-crop and re-save the fields in something like FIJI, which is also slow and not great for data storage.

As example software, SMAP can do this. It has a bit more complex system for working with dual view setups, but a simple option here to draw a rectangle or square on the image, or to define x0,y0,width,height or center,radius, would be sufficient.

straussmaximilian, 2022-01-29:
Hi,
when opening localize and loading data you can directly draw a rectangle and only the spots within will be processed. Is this what you are looking for?

[image]

GolgiWhillikers, 2022-02-03:
Ah...so it does. I was not aware of that feature, and it does work in my version, thanks.

One thing I have noticed with drawing the box is that it's not limited to the actual image bounds, so if you accidentally select some white space outside the image, Picasso will try to localize out there. This has come up while using it to localize say, the left half of a dual view image, where I'm selecting a large space that goes right to the image edges. Is it possible to limit the box to the image bounds? Though, it's not a major problem with usage.

Is it possible to add this to the command line localize call too, so you could call a box to localize within from cmd?

closed 2022-09-28 by rafalkowalewski1