jungmannlab/picasso #239: Wish: Looking at fit parameters

ajasja, 2022-05-03:
Picasso version: 0.4 (pip)
Python version: 3.8.5
Operating System: Win 10

### Description

I'm looking for the best way to filter the fitted localizations. I would like to click on a fit and see the parameters. Is there an easy a way to do this?

### What I Did

Localized and fitted a tiff file. I'm getting several spurious fits that should be easy to filer away, but I don't have a feeling for which fit parameters are best to filter on.  
[image]

If I could click on a localization and see the fit parameters that would be great.

cleterrier, 2022-05-03:
Here it looks like a simple filtering on x and Y coordinates (being close to the edge) would catch all these unwanted localizations?

ajasja, 2022-05-04:
@cleterrier indeed, that's what I'm doing now, but there are still spurious fits in other frames in the middle of the image.

ajasja, 2022-05-04:
It seems like the FitMarker (https://github.com/jungmannlab/picasso/blob/ac0064fbc232caab1135ebd3c4cc55f439a1cd32/picasso/gui/localize.py#L158) could be extended with mouse over or click events relatively easily.

ajasja, 2022-06-16:
@rafalkowalewski1 Thanks! I hope I solved the problem by only looking at spots near the fibers, but I still think it would be cool if you could click on the fit and see the parameters