jungmannlab/picasso #258: Pip installation problems M1 macOS

steinjohannes, 2022-07-22:
* Picasso version: 76a1ead4c224090f80a6af1662a122998e9dfe99
* Python version: 3.8
* Operating System: M1 macOS Monterey

### Description

Following the pip installation guide and after apparent succesful installation, I receive the following error code when calling a picasso module (e.g. localize): ModuleNotFoundError: No module named 'picasso.ext'

### What I Did

- conda create —name picasso_sr python=3.8
- conda activate picasso_sr
- conda install -c anaconda pyqt 
- pip install picassosr
- picasso localize

```
(base) wyssuser@wyss12422 ~ % conda create --name picasso_sr python=3.8 
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /Users/wyssuser/opt/anaconda3/envs/picasso_sr

  added / updated specs:
    - python=3.8

The following NEW packages will be INSTALLED:

  ca-certificates    pkgs/main/osx-64::ca-certificates-2022.4.26-hecd8cb5_0
  certifi            pkgs/main/osx-64::certifi-2022.6.15-py38hecd8cb5_0
  libcxx             pkgs/main/osx-64::libcxx-12.0.0-h2f01273_0
  libffi             pkgs/main/osx-64::libffi-3.3-hb1e8313_2
  ncurses            pkgs/main/osx-64::ncurses-6.3-hca72f7f_3
  openssl            pkgs/main/osx-64::openssl-1.1.1q-hca72f7f_0
  pip                pkgs/main/osx-64::pip-22.1.2-py38hecd8cb5_0
  python             pkgs/main/osx-64::python-3.8.13-hdfd78df_0
  readline           pkgs/main/osx-64::readline-8.1.2-hca72f7f_1
  setuptools         pkgs/main/osx-64::setuptools-61.2.0-py38hecd8cb5_0
  sqlite             pkgs/main/osx-64::sqlite-3.38.5-h707629a_0
  tk                 pkgs/main/osx-64::tk-8.6.12-h5d9f67b_0
  wheel              pkgs/main/noarch::wheel-0.37.1-pyhd3eb1b0_0
  xz                 pkgs/main/osx-64::xz-5.2.5-hca72f7f_1
  zlib               pkgs/main/osx-64::zlib-1.2.12-h4dc903c_2

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate picasso_sr
#
# To deactivate an active environment, use
#
#     $ conda deactivate

(base) wyssuser@wyss12422 ~ % conda activate picasso_sr
(picasso_sr) wyssuser@wyss12422 ~ % conda install -c anaconda pyqt
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /Users/wyssuser/opt/anaconda3/envs/picasso_sr

  added / updated specs:
[... 389 lines cut ...]
Collecting webencodings
  Downloading webencodings-0.5.1-py2.py3-none-any.whl (11 kB)
Collecting pycparser
  Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 118.7/118.7 kB 3.6 MB/s eta 0:00:00
Building wheels for collected packages: validators
  Building wheel for validators (setup.py) ... done
  Created wheel for validators: filename=validators-0.20.0-py3-none-any.whl size=19582 sha256=fc43eb5d36343489997de6fc86d8704d96744034d3d59c54940c337c73eba633
  Stored in directory: /Users/wyssuser/Library/Caches/pip/wheels/19/09/72/3eb74d236bb48bd0f3c6c3c83e4e0c5bbfcbcad7c6c3539db8
Successfully built validators
Installing collected packages: webencodings, wcwidth, Send2Trash, pytz, PyQt5-Qt5, pure-eval, ptyprocess, pickleshare, mistune, ipython-genutils, fastjsonschema, executing, commonmark, backcall, appnope, zipp, watchdog, urllib3, tzdata, typing-extensions, traitlets, tqdm, tornado, toolz, toml, tinycss2, threadpoolctl, tenacity, soupsieve, smmap, six, semver, pyzmq, pyyaml, pyrsistent, PyQt5-sip, pyparsing, pympler, pygments, pycparser, psutil, protobuf, prompt-toolkit, prometheus-client, pillow, pexpect, parso, pandocfilters, numpy, nest-asyncio, MarkupSafe, llvmlite, kiwisolver, jupyterlab-widgets, jupyterlab-pygments, joblib, idna, greenlet, future, entrypoints, defusedxml, decorator, debugpy, cycler, colorama, click, charset-normalizer, cachetools, blinker, backports.zoneinfo, attrs, asteval, validators, uncertainties, terminado, sqlalchemy, scipy, rich, requests, pytz-deprecation-shim, python-dateutil, PyQt5, pyarrow, plotly, patsy, packaging, numba, matplotlib-inline, jupyter-core, jinja2, jedi, importlib-resources, importlib-metadata, h5py, gitdb, cffi, bleach, beautifulsoup4, asttokens, tzlocal, stack-data, scikit-learn, pandas, matplotlib, lmfit, jupyter-client, jsonschema, gitpython, argon2-cffi-bindings, statsmodels, nbformat, ipython, argon2-cffi, altair, plotly-express, nbclient, ipykernel, nbconvert, notebook, widgetsnbextension, ipywidgets, pydeck, streamlit, picassosr
Successfully installed MarkupSafe-2.1.1 PyQt5-5.15.7 PyQt5-Qt5-5.15.2 PyQt5-sip-12.11.0 Send2Trash-1.8.0 altair-4.2.0 appnope-0.1.3 argon2-cffi-21.3.0 argon2-cffi-bindings-21.2.0 asteval-0.9.27 asttokens-2.0.5 attrs-21.4.0 backcall-0.2.0 backports.zoneinfo-0.2.1 beautifulsoup4-4.11.1 bleach-5.0.1 blinker-1.5 cachetools-5.2.0 cffi-1.15.1 charset-normalizer-2.1.0 click-8.1.3 colorama-0.4.4 commonmark-0.9.1 cycler-0.11.0 debugpy-1.6.2 decorator-5.1.1 defusedxml-0.7.1 entrypoints-0.4 executing-0.8.3 fastjsonschema-2.16.1 future-0.18.2 gitdb-4.0.9 gitpython-3.1.27 greenlet-1.1.2 h5py-3.7.0 idna-3.3 importlib-metadata-4.12.0 importlib-resources-5.8.0 ipykernel-6.15.1 ipython-8.4.0 ipython-genutils-0.2.0 ipywidgets-7.7.1 jedi-0.18.1 jinja2-3.1.2 joblib-1.1.0 jsonschema-4.7.2 jupyter-client-7.3.4 jupyter-core-4.11.1 jupyterlab-pygments-0.2.2 jupyterlab-widgets-1.1.1 kiwisolver-1.4.4 llvmlite-0.38.1 lmfit-1.0.1 matplotlib-3.3.2 matplotlib-inline-0.1.3 mistune-0.8.4 nbclient-0.6.6 nbconvert-6.5.0 nbformat-5.4.0 nest-asyncio-1.5.5 notebook-6.4.12 numba-0.55.2 numpy-1.22.4 packaging-21.3 pandas-1.4.3 pandocfilters-1.5.0 parso-0.8.3 patsy-0.5.2 pexpect-4.8.0 picassosr-0.4.6 pickleshare-0.7.5 pillow-9.2.0 plotly-5.9.0 plotly-express-0.4.1 prometheus-client-0.14.1 prompt-toolkit-3.0.30 protobuf-3.20.0 psutil-5.8.0 ptyprocess-0.7.0 pure-eval-0.2.2 pyarrow-8.0.0 pycparser-2.21 pydeck-0.7.1 pygments-2.12.0 pympler-1.0.1 pyparsing-3.0.9 pyrsistent-0.18.1 python-dateutil-2.8.2 pytz-2022.1 pytz-deprecation-shim-0.1.0.post0 pyyaml-5.3.1 pyzmq-23.2.0 requests-2.28.1 rich-12.5.1 scikit-learn-0.23.2 scipy-1.8.1 semver-2.13.0 six-1.16.0 smmap-5.0.0 soupsieve-2.3.2.post1 sqlalchemy-1.4.22 stack-data-0.3.0 statsmodels-0.13.2 streamlit-1.10.0 tenacity-8.0.1 terminado-0.15.0 threadpoolctl-3.1.0 tinycss2-1.1.1 toml-0.10.2 toolz-0.12.0 tornado-6.2 tqdm-4.50.2 traitlets-5.3.0 typing-extensions-4.3.0 tzdata-2022.1 tzlocal-4.2 uncertainties-3.1.7 urllib3-1.26.10 validators-0.20.0 watchdog-2.1.4 wcwidth-0.2.5 webencodings-0.5.1 widgetsnbextension-3.6.1 zipp-3.8.1

(picasso_sr) wyssuser@wyss12422 ~ % picasso localize
Traceback (most recent call last):
  File "/Users/wyssuser/opt/anaconda3/envs/picasso_sr/bin/picasso", line 8, in <module>
    sys.exit(main())
  File "/Users/wyssuser/opt/anaconda3/envs/picasso_sr/lib/python3.8/site-packages/picasso/__main__.py", line 1530, in main
    from picasso.gui import localize
  File "/Users/wyssuser/opt/anaconda3/envs/picasso_sr/lib/python3.8/site-packages/picasso/gui/localize.py", line 19, in <module>
    from .. import io, localize, gausslq, gaussmle, zfit, lib, CONFIG, avgroi
  File "/Users/wyssuser/opt/anaconda3/envs/picasso_sr/lib/python3.8/site-packages/picasso/io.py", line 23, in <module>
    from .ext.bitplane import IMSWRITER
ModuleNotFoundError: No module named 'picasso.ext'
```

straussmaximilian, 2022-07-22:
Could you check if the error also occurs when you start picasso with `python -m picasso localize`?
In any case, I just pushed a new version to PyPi that should fix the underlying import error. Please comment in case this works/does not.

steinjohannes, 2022-07-23:
Hi Max, 

thanks for your quick response. I still received the same error regardless of wether using '''python -m picasso localize''' or '''picasso localize'''. I checked the picasso package content '''/Users/wyssuser/opt/anaconda3/envs/picasso_sr/lib/python3.8/site-packages/picasso''' and found that indeed the folder 'ext' containing bitplane.py was missing. Copying it in from the current repo fixed the issue. I also checked at [https://pypi.org/project/picassosr/](https://pypi.org/project/picassosr/), it's also missing here. I guess running the builder once more should fix that?

straussmaximilian, 2022-08-05:
Hi,
I did publish an updated version to PyPi again. Could you try if it works (`pip install picassor --upgrade`) now?

steinjohannes, 2022-08-09:
Works, thanks Max!

closed 2022-09-27 by rafalkowalewski1