jungmannlab/picasso #266: Inconsistent number of localisations in locs and spots files

mb1069, 2022-09-23:
* Picasso version: 0.4.11
* Python version: 3.8.5
* Operating System: Ubuntu & MacOS

### Description

Running localisation of spots functions, but hdf5 files contain different number of rows / images in stacks.

### What I Did
Ran localize on this tif file, 
[NPC-A647-3D-BEADS-as-stacks.tif.zip](https://github.com/jungmannlab/picasso/files/9635577/NPC-A647-3D-BEADS-as-stacks.tif.zip)

Localisation output and spots can be found here;
[Archive.zip](https://github.com/jungmannlab/picasso/files/9635592/Archive.zip)

rafalkowalewski1, 2022-09-26:
Hi Miguel
I took a closer look into your data and both the files in the archive folder have 94 spots/localizations (I read them with h5py). 
Moreover, the metadata files (.yaml) say that there is only one frame included, which is strange as the .tif file contains 41 frames.
Also, I can see that you used min net gradient of 2500. But when I try reproducing your results, I only get 34 spots using such parameters.
Are you sure you used the same files?

PS. I would not recommend using such a large box size for 3d calibration. Well calibrated setup should work well with the box size of 11 or maybe 13 at most. In fact, I would go for frames 10-35 only since the first and last frames show very blurred spots. You do not need such densly packed beads either : )

Please let me know if that was of any help and ask feel free to ask me more questions!

mb1069, 2022-09-26:
Hi,
Thanks for getting back to me,
Sorry about my example files, I must have gotten something mixed up.
I've re-localised the file and found the same error again.
The locs table contains 2978 rows, but the spots h5py contains 2979 spots.
[proof.zip](https://github.com/jungmannlab/picasso/files/9644869/proof.zip)

Thank you for the advice regarding the box size, I'm giving that a try ASAP!

rafalkowalewski1, 2022-09-26:
Hello again,

This time everything matches your results. The reason why you lose one spot is the sanity check that is run before saving localizations. It simply checks that all locs are finite (i.e., non-infinite or NaN), if x and y coordinates are between 0 and the width/height of your FOV and if localization precision is positive. I would guess that this one localization that was discarded had its localization precision negative (due to some error in fitting).

Again, if there are any other questions, I will be happy to assist!

mb1069, 2022-09-26:
Perfect, I thought something like that might have been happening.
Is there a way of ensuring the ordering / count of the spots matches the locs table?

rafalkowalewski1, 2022-09-26:
The only solution I can come up with is to alter the function resposible for the sanity check. You can do that by modifying the function "ensure_sanity" in picasso/lib.py. 
Keep in mind that you will get errors if you try to render localizations with negative localization precision. This is why I am reluctant to change the official version of picasso.

mb1069, 2022-09-26:
How about doing the opposite by removing the spot which doesn't have a corresponding entry in the localisation table?

rafalkowalewski1, 2022-09-27:
I could try to incorporate that in the future but that could only work when spots are localized.

closed 2022-09-28 by rafalkowalewski1

mb1069, 2024-07-30:
@rafalkowalewski1 Do you know if the order of spots corresponds to the order in the localisations table? I think I might have occasional mis-matches when extracting the data manually

rafalkowalewski1, 2024-08-01:
Hi, I think the order should be the same but there will be missing data in the localizations list since some spots are discarded as they yield localizations that are filtered (for example, outside the FOV).