jungmannlab/picasso #270: QObject::setParent: Cannot set parent, new parent is in a different thread

IrrelevantNameTom, 2022-10-27:
* Picasso version: 0.4.11
* Python version: 3.10.6 and 3.8.13
* Operating System: windows10

Hi,
Whether I use the standalone or install through pip the picasso localize freezes when launching Identify or Localize.
A windows "An error occured" pops up but without any error message. Poweshell precises "QObject::setParent: Cannot set parent, new parent is in a different thread"

I am reading a 16bit tif file (on computer not server). I tried different sizes (10, 100 and 1000 frames)

I also have an warning when launching localize:
"picasso\__init__.py:16: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  CONFIG = _yaml.load(config_file)"

But I dont think its related.
Thanks : )
Tom

straussmaximilian, 2022-10-31:
Hi,

This sounds like a bug.
The yaml warning is unrelated.
Could you share the file that caused you problems to investigate?

IrrelevantNameTom, 2022-11-01:
Hi,

Thank you for your answer. Here is the file we try to read. 
[Measurement1bin10.zip](https://github.com/jungmannlab/picasso/files/9912627/Measurement1bin10.zip)

This is a .tif file, 16bit. I did not find test file in the package, do you find you could share a valid one so I can exclude it comes form my format ?

Thank you : )
Tom

rafalkowalewski1, 2022-11-02:
Hey Tom,

I successfully opened and localized your file and I cannot reproduce your issue unfortunately. Can you please try identifying/localizing with this dataset?
[test.ome.tif.zip](https://github.com/jungmannlab/picasso/files/9918747/test.ome.tif.zip)

Bests,
Rafal

IrrelevantNameTom, 2022-11-02:
Hi Rafal,

Thanks for the test data. The same error happened unfortunately. Here is a screenshot of the error. I will try to install it on another computer and let you know if this is working better. 

Tom
[image]

IrrelevantNameTom, 2022-11-03:
Update: I installed the standalone Picasso on another computer and everything worked ok. The bug probably comes from my current computer. I will tell you if I can solve this.

Tom

rafalkowalewski1, 2022-11-03:
Hey Tom, thanks for the updates!

closed 2022-11-08 by rafalkowalewski1