jungmannlab/picasso #271: Opening .tifs not acquired from uManager

KDBuchan, 2022-11-03:
* Picasso version: 0.4.11
* Python version: 3.8.10
* Operating System: Windows 10 Home

### Description

I am trying my best to get open source PALM working with my data sets as I find Nikon's localisation software a little limiting. Currently I am not acquiring my TIFFs in micromanager as we use a core facility and I'm the only person doing this. 

### What I Did
After acquiring as nd2 files and converting to a tiff file or ome tiff and loading into localise I get this error for both.

[image]

If I load either of these into FIJI and export as a .raw file I get this error.

[image]

Can I get any help with this. Would love to use your software.

Cheers!
Kyle

rafalkowalewski1, 2022-11-05:
Hi Kyle,

If I understood correctly, you would like to localize .nd2 files. If that's the case, I have greate news since this feature will ne included in the next picasso update (hopefully within this month).

Please let me know if that should help you out.
Bests,
Rafal

rafalkowalewski1, 2022-11-05:
PS. If you would like to use this feature already, feel free to try out the development branch of out github rep : )

KDBuchan, 2022-11-07:
> 

Hi Rafal!

Yes! This is exactly what i'd like to do and would simplify things immensely! I'd be happy to try it out - unfortunately i'm a bit of a Github noob - where can I find the files/instructions for getting nd2 files to work? I'm not sure where the development branch is. D:

Thank you so much! 
Kyle

rafalkowalewski1, 2022-11-07:
Hi Kyle,

I will try to get a new picasso update early next week. If you do not want to wait, please follow this procedure to use the development branch on your computer:

1. I assume you're using Windows and that you have anaconda installed, if that's not the case, please let me know
2. Open anaconda prompt and go to the directory where you would like to store your local picasso
3. Type: git clone -b development https://github.com/jungmannlab/picasso.git
4. After the command is exectured, type: cd picasso
5. Then, you will need to create a new environment; to do that, type: conda create --name picasso_dev python=3.8
6. Follow the instructions in the command window 
7. Once the environment is created, activate it by typing: conda activate picasso_dev
8. Then, you need to install all packages. Type: pip install -r requirements.txt
9. After a longer time (around 20 minutes), your picasso development is ready
10. To open Localize, type: python -m picasso localize
11. if you want to open any other app, change localize to render, filter, etc

KDBuchan, 2022-11-08:
Hey Rafal,

Thanks for your help, I really appreciate it! It seems that I got the development branch installed okay, so I opened localise and dragged my .nd2 file into the window, and got the following error:

[image]

Any advice on how to proceed? :)

Thanks again!
Kyle

Heerpa, 2022-11-09:
Hi Kyle,
nikon saves a lot of metadata into their files, depending on the exact acquisition and hardware configuration. Probably, your metadata describes a configuration we haven't seen yet. Would it be possible to share your file with us, so we can investigate the difference in metadata structure?

KDBuchan, 2022-11-10:
> Hi Kyle, nikon saves a lot of metadata into their files, depending on the exact acquisition and hardware configuration. Probably, your metadata describes a configuration we haven't seen yet. Would it be possible to share your file with us, so we can investigate the difference in metadata structure?

Hi there! I'd be happy to. I will first trim the file to a few hundred frames and send it over, so it's not so big. I will do so in the next day or so. Thanks!

KDBuchan, 2022-11-11:
Hi folks,

It seems I can't share .nd2 files here in the forum, is there another way I can send this file to you?

Thank you!

rafalkowalewski1, 2022-11-12:
How about you try compressing your file to .zip or .rar?

Heerpa, 2022-11-13:
Or could you send a we-transfer link, or something of the sort provided by your institution? Google didn't narrow down closer than University of Texas Health Science Center, not to a lab page.

rafalkowalewski1, 2022-11-23:
Hey Kyle,

Sorry for the long waiting time. We have worked on your problem and it should be resolved in the next picasso release (this or the next week)

Bests,
Rafal

closed 2022-11-23 by rafalkowalewski1