jungmannlab/picasso #294: Render Clustering files output no more '_dbscluster' information

yunqingli62, 2022-11-25:
* Picasso version: v0.5.4
* Python version: 3.8.12
* Operating System: windows 10

### Description
The implement clustering algorithms only save a  '_clustered.hdf5' file, which is similar to the old 'dbscan.hdf5' file. But the old '_dbcluster.hdf5' file, and the informationin it are totally missing. 

### What I Did
I used the one-click installer installed the new version of Picasso and opened Render from its desktop icon, loaded my linked file and runed the clustering algorithm implemented in Render.

rafalkowalewski1, 2022-11-25:
Hi,

The way clustering is performed has been significantly changed in the latest release (0.5). If you would like to use the cluster information, please check the box "Save cluster centers" in the dialog asking for dbscan parameters. Also slightly different things are calculated in the cluster_centers files. 

I hope this helps, please let me know if you have any further questions.

Bests,
Rafal

yunqingli62, 2022-11-25:
Hi,

thanks a lot for the rapid response!
I tried the "save cluster centers" as well. Then beside the '_clustered' file I got another set of '_clustered_cluster_centers' file. Again, it shoulds the same coloums as the 'clustered' file. But it does not help with the information 'groups', 'convex_hull', 'area', 'mean_frame', 'std_frame' usw.
Especially, I think a lot of DNA-PAINT users rely on filtering 'mean_frame' and 'std_frame' to controll our data quality.
If it's possible, I really want to know how to get the old '_dbcluster' information.

Bests,
Yunqing

rafalkowalewski1, 2022-11-25:
Dear Yunqing,

I will retrieve some of these features in the next release (probably next week).

Bests,
Rafal

closed 2022-11-25 by rafalkowalewski1

yunqingli62, 2022-11-25:
Dear Rafal,

that's great!
Thanks a looooot!

Bests,
Yunqing