jungmannlab/picasso #295: Error opening TIF file

cleterrier, 2022-11-28:
* Picasso version: 0.5.4
* Python version: 3.8
* Operating System: Windows 10

I installed Picasso (using the conda method and pip -install -picassosr) 
I also installed GPUFit
I launched the Localize module
I used the File>Open Movie command on the following tif file (bead Z-stack for Z calibration, 256x256x160, size 20 MB):
https://filesender.renater.fr/?s=download&token=39a1f9b7-ca4f-489f-aa6c-f43ef98abf34

I got the following error:

```
Error in sys.excepthook:
Traceback (most recent call last):
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\gui\localize.py", line 2034, in excepthook
    errorbox.exec_()
AttributeError: 'StandardButton' object has no attribute 'exec_'

Original exception was:
Traceback (most recent call last):
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\gui\localize.py", line 1210, in open_file_dialog
    self.open(path)
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\gui\localize.py", line 1220, in open
    result = io.load_movie(path, prompt_info=prompt_info)
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\io.py", line 189, in load_movie
    return load_tif(path)
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\io.py", line 174, in load_tif
    movie = TiffMultiMap(path, memmap_frames=False)
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\io.py", line 951, in __init__
    matches = [_re.match(pattern, _) for _ in entries]
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\io.py", line 951, in <listcomp>
    matches = [_re.match(pattern, _) for _ in entries]
NameError: free variable 'pattern' referenced before assignment in enclosing scope
```

rafalkowalewski1, 2022-11-28:
Hi Christophe,

I suspect the issue is raised by an incorrect file format. Picasso Localize operates on .ome.tif  and NDTiffStack files. Did you record the attached movie in MicroManager?

Bests,
Rafal

cleterrier, 2022-11-28:
Hi, it's a tif file produced from an nd2 file via BioFormat in Fiji. Of note, an OME-TIF file exported by BioFormat from Fiji has the same error. I can't directly use the nd2 file as I need to remove frames in Fiji before using it for 3D calibration.

rafalkowalewski1, 2022-11-29:
Hi Christophe,

I played around with the file and discovered that the simple solution to your problem is to manually rename your file to end with ".ome.tif" instead of ".tif" only. Localize was built to support this file format and recently .nd2 and NDTiffStack have been added, but other files are not supported.

I hope this helps.

Bests,
Rafal

closed 2022-11-29 by rafalkowalewski1