jungmannlab/picasso #305: error in dbscan cmd

Z-Qing, 2022-12-14:
* Picasso version: 0.5.6
* Python version:3.8.14
* Operating System:win11

## description
The command picasso dbscan asks for arguement pixelsize. But only file path, radius, density are mentioned in the help file.

## What I did
I run dbscan using cmd. The cmd I typed is "picasso dbscan path 1 100"

And I get the following error message:
Traceback (most recent call last):
  File "D:\Anaconda3\envs\picasso\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Anaconda3\envs\picasso\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\Anaconda3\envs\picasso\Scripts\picasso.exe\__main__.py", line 7, in <module>
  File "D:\Anaconda3\envs\picasso\lib\site-packages\picasso\__main__.py", line 1619, in main
    _dbscan(args.files, args.radius, args.density)
  File "D:\Anaconda3\envs\picasso\lib\site-packages\picasso\__main__.py", line 454, in _dbscan
    clusters, locs = postprocess.dbscan(locs, radius, min_density)
TypeError: dbscan() missing 1 required positional argument: 'pixelsize'

rafalkowalewski1, 2022-12-15:
Hello,

Thank you for raising the issue. I will fix it in the next Picasso release. If this is fine with you, you can use the following alternative in Render for now: Load the file you would like to cluster, click Postprocess in the menu bar, click Clustering > DBSCAN. The window with DBSCAN paramters will open. 

Bests,
Rafal

Z-Qing, 2022-12-15:
Thanks for the reply. I will just call postprocess.dbscan() function to work around this. 
I have a separate question. Can I call the function localize.localize() and put parameters "info"  and "parameters" manually to get localizations? Will this be any faster than using the localize cmd?

rafalkowalewski1, 2022-12-17:
It should work, however, the gain in speed shouldn't be significant. Also, calling localize.localize() uses MLE fitting which is much slower that LQ gaussian fitting.
I hope this helps.
Bests,
Rafal

closed 2022-12-18 by rafalkowalewski1