jungmannlab/picasso #308: Failed installation in mac M1

agvesga, 2023-01-25:
* Picasso version: 0.5.0
* Python version: 3.8
* Operating System: mac M1

### Description

While trying to install picasso in my mac with chip Apple M1 Pro I get the same error over and over.

### What I Did

I tried to reinstall numpy and used the version 0.4.0 of picasso, with the same result.

```
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: legacy-install-failure
      
      × Encountered error while trying to install package.
      ╰─> numpy
      
      note: This is an issue with the package mentioned above, not pip.
      hint: See above for output from the failure.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

agvesga, 2023-01-26:
Hi again, 

I managed installing it by setting the requirements.txt file:

> PyQt5
> numpy>=1.19.2
> h5py>=2.10.0
> matplotlib==3.3.2
> numba>=0.51.2
> 
> **scipy>=1.6.2**
> 
> pandas==1.3.0
> pyyaml==5.3.1 
> 
> **scikit-learn>=0.23.2**
> 
> colorama==0.4.4
> tqdm==4.50.2
> lmfit==1.0.1
> streamlit==1.10.0
> moviepy==1.0.3
> nd2==0.3.0
> sqlalchemy==1.4.22
> plotly-express==0.4.1
> watchdog==2.1.4
> psutil==5.8.0
> protobuf==3.20
> nd2reader==3.3.0

Nevertheless I don't know whether it will have full functionality. For instance, the render does not seem to work properly

rafalkowalewski1, 2023-01-27:
Hello,

I recommend installing picasso using pip (run the following command):
pip install picassosr
This approach has been tested and should work.

I recommend creating a separate conda environment for this, you can find instructions on the main repository page.

Please let me know if there is something else I could help you with.

rafalkowalewski1, 2023-01-27:
PS. picasso 0.5.0 is not the most up to date version, I also recommend using the latest versions since they usually have fewer bugs : )

agvesga, 2023-01-27:
Hello, 

thanks for the advice. I have to say that scipy==1.6.2 is not built for M1 (at least it is not in the main repositories). Had to go to scipy page in github, download and install the correct version by hand. Also had to install numpy==1.23.0, since 1.19.2 is not supported for M1 either.

But for now, everything runs.

Also, I have another question. The samples in https://github.com/jungmannlab/picasso/blob/master/samples/SampleNotebook.ipynb are referring to a file 'testdata_locs.hdf5', but there is no such a file in your repo, right?

Thanks again!

rafalkowalewski1, 2023-01-30:
Hi,

The .hdf5 file is not in repo, however, you can generate it by using Localize on tests/testdata.raw

I will work on improving the sample notebooks in the near future, they have not been updated for some time : )

Bests,
Rafal

closed 2023-02-07 by rafalkowalewski1